alexa-pi/AlexaPi #299: Alexa not booting up after running setup.sh

Jojeker, 2017-08-22:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian
Linux raspberry 4.9.39-v7+ #1021 SMP Mon Jul 24 11:32:18 BST 2017 armv7l GNU/Linux
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi
```

**Python release (`python2 --version`):**
```
Python 2.7.9          
```

**Description of problem:**
```
Alexa Pi not booting up. Description of error:

root@raspberry:/opt# sudo systemctl status AlexaPi.service -l
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Di 2017-08-22 17:32:53 UTC; 6min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 17620 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 17620 (code=exited, status=1/FAILURE)

Aug 22 17:32:53 raspberry systemd[1]: Started Alexa client for all your devices.
Aug 22 17:32:53 raspberry python[17620]: Traceback (most recent call last):
Aug 22 17:32:53 raspberry python[17620]: File "/opt/AlexaPi/src/main.py", line 16, in <module>
Aug 22 17:32:53 raspberry python[17620]: from future.builtins import bytes
Aug 22 17:32:53 raspberry python[17620]: ImportError: No module named future.builtins
Aug 22 17:32:53 raspberry systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Aug 22 17:32:53 raspberry systemd[1]: Unit AlexaPi.service entered failed state.

```

**Expected:**
```
Boot up after first install.
```

**Problem-relevant `config.yaml` entries:**
```
As long as I know everything fine. Here is the config.yaml:

# See https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging
# to get a better idea of how to set things in here

# Level of log output
# default: INFO
# set to DEBUG for verbose logging
logging: INFO

sound:
  # Name of your ALSA microphone/soundcard in arecord -L
  input_device: "plughw:CARD=Device,DEV=0"
  # This is here if none of the listed device: "raspberrypi"
  # This is an override that is normally not needed and is a last resort.
  allow_unlisted_input_device: false

  playback_handler: "sox"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: "alsa"
  # output device: "raspberrypi"
  # currently supports ALSA only
  output_device: "default"

  # if left empty, won't be passed to the underlying library / command if it supports it
  # if it doesn't 100% will be used
  default_volume: 80

  # Means the same as above, but is used for media (playing music for example)
  # Leave blank to use the same setting as for speech content
  media_output:
  media_output_device:
  media_default_volume:

  # Some device: "raspberrypi"
  # value is the number of seconds of silence to pad the beginning and end of the audio. Only
  # the sox handler supports this configuration option.
  playback_padding: 0

# Amazon Alexa settings
alexa:
 Client_ID: "Filled Out"
 Client_Secret: "Filled Out"
  Device_Type_ID: "Filled Out"
  Security_Profile_Description: "Too"
  Security_Profile_ID: "Too"
  refresh_token: "Too"

# See https://github.com/alexa-pi/AlexaPi/wiki/Triggers for the description of these options
triggers:
  platform:
    enabled: true
    voice_confirm: false

    # one of: oneshot-vad, continuous, continuous-vad
    event_type: "oneshot-vad"

    # only for "continuous" event_types
    long_press:
      # system command to run after pressing the button for _duration_ number of seconds
      # empty to disable
      command: ""

      # duration in seconds
      duration: 10

      # CURRENTLY DISABLED
      # optional audio file to play before executing the command
      # you might wanna use full path to make sure AlexaPi finds it
      # you can use {resources_path} for the path to the AlexaPi "resources" directory
      audio_file: ""

  pocketsphinx:
    enabled: true
    voice_confirm: true

    phrase: "alexa"
    threshold: 1e-10

# Commands to run before and after an interaction begins. Leave empty to disable.
event_commands:
event_commands:

  # Example used to lower shairport-sync volume during interaction with alexa
  #     startup: "amixer set shairportvolume 100%"
  #     pre_interaction: "amixer set shairportvolume 10%-""
  #     post_interaction: "amixer set shairportvolume 10%+""
  # Note this assumes you have redirected shairport-syncs audio to a seperate softdevice
  # with control "shairportvolume". Example asound.conf avalible in the scripts folder

  # Command to run at the start of AlexaPi
  startup: ""

  # Command to run after a trigger is detected but before an interaction begins
  pre_interaction: ""

  # Command to run after the interaction has completed
  post_interaction: ""

  # Command to run during a clean shutdown of AlexaPi
  shutdown: ""

platform:
  # Name of your platform, e.g. raspberrypi, orangepi, desktop
  device: "raspberrypi"

platforms:
  # common config for all platforms (if they support the options)
  # can be overriden in the the respective platform sections
  common:

  chip:
    # GPIO Pin with button connected
    button: XIO-P1
    # GPIO Pin for the playback/activity light
    plb_light: XIO-P3
    # GPIO Pin for the recording light
    rec_light: XIO-P5

  raspberrypi:
    # GPIO Pin with button connected
    button: 18
    # GPIO Pin for the playback/activity light
    plb_light: 24
    # GPIO Pin for the recording light
 rec_light: 25

  orangepi:
    # GPIO Pin with button connected
    button: 20
    # GPIO Pin for the playback/activity light
    plb_light: 9
    # GPIO Pin for the recording light
    rec_light: 8

  desktop:
    min_seconds_to_record: 3

  dummy:

  magicmirror:
    # Magic Mirror - github.com/MichMich/MagicMirror
    # Integrates with MMM-AlexaPi module (github.com/dgonano/MMM-AlexaPi) to display listeni$
    # Hostname of the AlexaPi HTTP server, leave empty for any name
    hostname: ""
    # Port the ALexaPi HTTTP server binds to
   port: 5055
    # Hostname of the Magic Mirror
    mm_hostname: "localhost"
    # Port of the Magic Mirror
    mm_port: "8080"
    # How ofter the Hearbeat fires in seconds
    hb_timer: 5

  hyperion:
    # Hostname of the Hyperion node's JSON server
    hyperion_json_host: "127.0.0.1"
    # Port of the Hyperion node's JSON server
    hyperion_json_port: 19444
    # Visual status as static 'color' or 'effect'
    # Overridden by 'hyperion_flash_state_xxx'
    hyperion_mode: "effect"
    # Running speed when visualizing as 'effect'
    hyperion_effect_speed: 1
    hyperion_priority: 1
    color_playback: [80, 213, 35]
    color_processing: [35,169,213]
    color_recording: [168, 35, 213]
    flash_duration: 500
    flash_frequency: 8
    # Use Strobe effect instead of default effect or static color
    flash_state_playback: true
    flash_state_processing: false
    flash_state_recording: false
    # Visualize playback state
    indicate_playback: true
    #Also print json request/response when in debug mode
    verbose: false

  serial:
    port: "/dev/ttyACM0"
    baudrate: 9600
    messages:
      trigger: ""
      playback_start: "g"
      playback_end: "x"
      recording_start: ""
      recording_end: ""
      processing_start: ""
      processing_end: ""
      failure: ""
      success: ""
```

**Steps to reproduce:**

1. Remove all python and pip packages and run setup.sh
2. Start Alexa service
3. Voilà :(

**Traceback (if applicable):**
```
Nothing to add.
```

**Additional info:**
```
Nothing to add.
```

EmerickH, 2017-08-23:
This is strange but you just have to do:
```
sudo pip install future
```
If there's others missing packages, you also have to install:
### For master branch (default)
```
requests
CherryPy
python-vlc
pyalsaaudio
webrtcvad
pyyaml
pocketsphinx
coloredlogs
```

### For dev branch.
```
requests
CherryPy
python-vlc
pyaudio
webrtcvad
pyyaml
pocketsphinx
coloredlogs
```

Emerick

Jojeker, 2017-08-25:
Thank you I am trying it out right now you will get a response shortly. I think I am going to install all pip packages to get sure that no other requirements exist, to run AlexaPi in a right way.

Thank you! :+1:

Jojeker, 2017-08-25:
Tried every package out but everything seems to be fine. Not sure what to do now :/

pi@raspberry:~ $ sudo pip install future requests CherryPy python-vlc pyaudio webrtcvad pyyaml pocketsphinx coloredlogsRequirement already satisfied: future in /usr/local/lib/python2.7/dist-packages
Requirement already satisfied: requests in /usr/local/lib/python2.7/dist-packages                                                                                                               
Requirement already satisfied: CherryPy in /usr/local/lib/python2.7/dist-packages                                                                                                               
Requirement already satisfied: python-vlc in /usr/local/lib/python2.7/dist-packages                                                                                                             
Requirement already satisfied: pyaudio in /usr/local/lib/python2.7/dist-packages                                                                                                                
Requirement already satisfied: webrtcvad in /usr/local/lib/python2.7/dist-packages                                                                                                              
Requirement already satisfied: pyyaml in /usr/local/lib/python2.7/dist-packages                                                                                                                 
Requirement already satisfied: pocketsphinx in /usr/local/lib/python2.7/dist-packages                                                                                                           
Requirement already satisfied: coloredlogs in /usr/local/lib/python2.7/dist-packages                                                                                                            
Requirement already satisfied: certifi>=2017.4.17 in /usr/local/lib/python2.7/dist-packages (from requests)                                                                                     
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /usr/local/lib/python2.7/dist-packages (from requests)                                                                                  
Requirement already satisfied: idna<2.7,>=2.5 in /usr/local/lib/python2.7/dist-packages (from requests)                                                                                         
Requirement already satisfied: urllib3<1.23,>=1.21.1 in /usr/local/lib/python2.7/dist-packages (from requests)                                                                                  
Requirement already satisfied: six in /usr/local/lib/python2.7/dist-packages (from CherryPy)                                                                                                    
Requirement already satisfied: portend>=2.1.1 in /usr/local/lib/python2.7/dist-packages (from CherryPy)                                                                                         
Requirement already satisfied: cheroot>=5.2.0 in /usr/local/lib/python2.7/dist-packages (from CherryPy)                                                                                         
Requirement already satisfied: humanfriendly>=3.2 in /usr/local/lib/python2.7/dist-packages (from coloredlogs)                                                                                  
Requirement already satisfied: tempora>=1.8 in /usr/local/lib/python2.7/dist-packages (from portend>=2.1.1->CherryPy)                                                                           
Requirement already satisfied: monotonic; python_version == "2.6" or python_version == "2.7" or python_version == "3.0" or python_version == "3.1" or python_version == "3.2" in /usr/local/lib/python2.7/dist-packages (from humanfriendly>=3.2->coloredlogs)
Requirement already satisfied: pytz in /usr/local/lib/python2.7/dist-packages (from tempora>=1.8->portend>=2.1.1->CherryPy)

I am still getting the same error:

pi@raspberry:~ $ sudo systemctl restart AlexaPi.service
pi@raspberry:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Fr 2017-08-25 09:53:20 UTC; 6s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 2719 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 2719 (code=exited, status=1/FAILURE)

Aug 25 09:53:19 raspberry systemd[1]: Started Alexa client for all your devices.
Aug 25 09:53:20 raspberry python[2719]: Traceback (most recent call last):
Aug 25 09:53:20 raspberry python[2719]: File "/opt/AlexaPi/src/main.py", line 16, in <module>
Aug 25 09:53:20 raspberry python[2719]: from future.builtins import bytes
Aug 25 09:53:20 raspberry python[2719]: ImportError: No module named future.builtins
Aug 25 09:53:20 raspberry systemd[1]: AlexaPi.service: main process exited, code=exited...URE
Aug 25 09:53:20 raspberry systemd[1]: Unit AlexaPi.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

Hope there is a simple solution for this problem.
Could it be possible because of the python version or other components to run AlexaPi properly?

Jojeker, 2017-08-25:
Short update -- starting main.py manuallly fixes the problem but over the service it is currently not working.

EmerickH, 2017-08-30:
Hello,
I think it's because you have two different versions of python.
Check your default version with `python --version`, install the correct pip version (pip2 or pip3 package) and use:
`sudo pip2 install future requests CherryPy python-vlc pyaudio webrtcvad pyyaml pocketsphinx coloredlogs`
or
`sudo pip3 install future requests CherryPy python-vlc pyaudio webrtcvad pyyaml pocketsphinx coloredlogs`

Emerick

renekliment, 2017-10-08:
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment