alexa-pi/AlexaPi #300: How to run on Pine64, getting module only supported on raspberrypi RPI_GPIO

dvdbot, 2017-08-23:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
AlexaPi fails to boot

**Your OS (including version) where you are running AlexaPi:**
```
ARMBIAN 5.31 stable Debian GNU/Linux 8 (jessie) 3.10.105-pine64
```

**Your hardware platform and model you are running on:**
```
Pine64 2GB
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
AlexaPi fails on boot cause unsupported module:
Log:
Aug 23 12:34:59 pine64 python[24574]: im = importlib.import_module('alexapi.device_platforms.' + config['plat...None)
Aug 23 12:34:59 pine64 python[24574]: File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
Aug 23 12:34:59 pine64 python[24574]: __import__(name)
Aug 23 12:34:59 pine64 python[24574]: File "/opt/AlexaPi/src/alexapi/device_platforms/raspberrypiplatform.py"...dule>
Aug 23 12:34:59 pine64 python[24574]: import RPi.GPIO as GPIO
Aug 23 12:34:59 pine64 python[24574]: File "/usr/local/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", lin...dule>
Aug 23 12:34:59 pine64 python[24574]: from RPi._GPIO import *
Aug 23 12:34:59 pine64 python[24574]: RuntimeError: This module can only be run on a Raspberry Pi!
Aug 23 12:34:59 pine64 systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Aug 23 12:34:59 pine64 systemd[1]: Unit AlexaPi.service entered failed state.

```

**Expected:**
```
AlexaPi runs
```

**Problem-relevant `config.yaml` entries:**
```
Wrong device probably, I've choosen RaspberryPi which is closest related to the Pine64 as far as I know
```

**Steps to reproduce:**

1. Install ARMBIAN on Pine64
2. follow these instructions on how to install AlexaPi: http://lifehacker.com/the-simplest-way-to-build-a-raspberry-pi-powered-amazon-1794218212
3. Try to start AlexaPi using following command: sudo systemctl start AlexaPi.service

**Traceback (if applicable):**
```
Aug 23 12:34:59 pine64 python[24574]: im = importlib.import_module('alexapi.device_platforms.' + config['plat...None)
Aug 23 12:34:59 pine64 python[24574]: File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_module
Aug 23 12:34:59 pine64 python[24574]: __import__(name)
Aug 23 12:34:59 pine64 python[24574]: File "/opt/AlexaPi/src/alexapi/device_platforms/raspberrypiplatform.py"...dule>
Aug 23 12:34:59 pine64 python[24574]: import RPi.GPIO as GPIO
Aug 23 12:34:59 pine64 python[24574]: File "/usr/local/lib/python2.7/dist-packages/RPi/GPIO/__init__.py", lin...dule>
Aug 23 12:34:59 pine64 python[24574]: from RPi._GPIO import *
Aug 23 12:34:59 pine64 python[24574]: RuntimeError: This module can only be run on a Raspberry Pi!
Aug 23 12:34:59 pine64 systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Aug 23 12:34:59 pine64 systemd[1]: Unit AlexaPi.service entered failed state.
```

**Additional info:**
```

```

EmerickH, 2017-08-23:
Hello,
Re-launch the install and choose dummy device instead.

Emeirkc

dvdbot, 2017-08-23:
@EmerickH I tried, dummy is a unknown option in the latest version, choose "Other" but couldn't get it to work with a 3.5mm microphone, found a USB webcam trying that atm on my RPI2b.

Another question, is it possible to use AlexaPi with a bluetooth speaker with built-in microphone?
Couldn't get autoconnect to work but couldn't get AlexaPi to use the bluetooth microphone (Alexa used the speaker) even though arecord does work with the bluetooth microphone

EmerickH, 2017-08-23:
@dvdbot Yes it's called Other in setup, sorry.
But the audio is not related at all with the platform.
You must [setup your audio](https://emerickh.github.io/AlexaPi-config/setup/alsa.html).

And of course you can use what you want as microphone/speaker.

dvdbot, 2017-08-23:
Tried that, didn't work, not even with the bluetooth speaker when it was connected...
arecord did work with the bluetoothspeakers microphone, will try again with the better documentation you just gave me

dvdbot, 2017-08-23:
I've got it running on the pi except the following error:
ALSAAudioError: Capture data too large. Try decreasing period size
How can I fix this?

dvdbot, 2017-08-23:
I tried changing the .setperiodsize() in "/opt/AlexaPi/src/alexapi/capture.py" and "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py" I increased and decreased but didn't work...
When using "arecord -r 16000 -c 1 -d 5 -f S16_LE -D plughw:1 test.wav" It just works and record

dvdbot, 2017-08-23:
Okay, I've got her to work with a Bluetooth speaker + mic but she isn't always responding, only responded like 2 times...
How can I solve this?

EmerickH, 2017-08-23:
I need the complete log to help you:
```
sudo service alexapi stop OR sudo systemctl stop AlexaPi.service
```
and
```
/opt/AlexaPi/src/main.py -d
```

Emerick

dvdbot, 2017-08-23:
Bluetooth broke, doing a reinstall asap...
I noticed changing the thresshold in config.yaml for Pocketphoenix helped solve this but didn't get it perfect but it was acceptable but somehow the bluetooth broke and I wasn't able to connect to the speaker + mic anymore

closed 2017-10-08 by renekliment