alexa-pi/AlexaPi #302: I can't find the device type id

billcui57, 2017-08-24:
Raspbian
Raspberry Pi 3

I can't find the device type id

 https://developer.amazon.com/avs/home.html#/avs/products/new
 I was provided with the security profile id description,Security Profile ID,Client ID,Client secret,Product ID,Security Profile Name,Security Profile Description, but developer.amazon.com did not show me the device type id, which was a requisite for the setup process. As a matter of fact, it didn't even prompt me to create a device type id at all! I looked through several installation tutorials, and all of them just glides over the fact that a device type id exists without telling me how to create one. Perhaps all the installation tutorials are outdated? Perhaps the amazon website was updated?

maso27, 2017-08-24:
It looks like you're right--amazon has updated their developer website.
What used to be called "Device Type ID" is now called "Product name".
It's the main name you've given your Alexa project.

Try that and see if it works.

Tardog, 2017-08-26:
A small correction: The former "Device Type ID" has been renamed to "Product ID", not "Product name".

closed 2017-08-26 by billcui57

Deranger72, 2017-12-25:
a very important correction Tardog thx ;-) - but you both saved my day - thx ;-)

inderpartap, 2018-04-02:
is it product ID or amazon ID?

iguana03, 2019-09-10:
not works Product ID neither Amazon ID

DjDemonD, 2020-10-20:
I just used Product ID from the page titled Alexa Voice Service - Products - Product details - Information - Product ID
Then the other fields which are named exactly the same on AlexaPi config as they are on Alexa Voice Service website.

rtaveras07, 2021-03-25:
is the name of your device, after asking by your OS....