alexa-pi/AlexaPi #303: Alexapi active:failed

billcui57, 2017-08-26:
Hey first of all thanks for reading my issue :)
I followed the installation process just fine, but when I ran 
`sudo systemctl status AlexaPi.service`
This showed up:

>  AlexaPi.service - Alexa client for all your devices
>    Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Sat 2017-08-26 13:09:57 UTC; 14min ago
>      Docs: https://github.com/alexa-pi/AlexaPi/wiki
>   Process: 821 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
>  Main PID: 821 (code=exited, status=1/FAILURE)
> 
> Aug 26 13:09:33 raspberrypi systemd[1]: Started Alexa client for all your devices.
> Aug 26 13:09:57 raspberrypi python[821]: CRITICAL: Your input_device 'pulse' is invalid. Use one of the following:
> Aug 26 13:09:57 raspberrypi python[821]: default
> Aug 26 13:09:57 raspberrypi python[821]: null
> Aug 26 13:09:57 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
> Aug 26 13:09:57 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
> Aug 26 13:09:57 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
> 
> 
> 

I investigated the issue by running
`sudo journalctl -u AlexaPi.service`
Which returned with:

> -- Logs begin at Thu 2016-11-03 17:16:43 UTC, end at Sat 2017-08-26 13:27:03 UTC. --
> Aug 26 13:09:07 raspberrypi systemd[1]: Started Alexa client for all your devices.
> Aug 26 13:09:19 raspberrypi python[426]: CRITICAL: Your input_device 'pulse' is invalid. Use one of the following:
> Aug 26 13:09:19 raspberrypi python[426]: default
> Aug 26 13:09:19 raspberrypi python[426]: null
> Aug 26 13:09:19 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
> Aug 26 13:09:19 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
> Aug 26 13:09:19 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
> Aug 26 13:09:33 raspberrypi systemd[1]: Started Alexa client for all your devices.
> Aug 26 13:09:57 raspberrypi python[821]: CRITICAL: Your input_device 'pulse' is invalid. Use one of the following:
> Aug 26 13:09:57 raspberrypi python[821]: default
> Aug 26 13:09:57 raspberrypi python[821]: null
> Aug 26 13:09:57 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
> Aug 26 13:09:57 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
> Aug 26 13:09:57 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
> 

I then stopped the service `sudo systemctl stop AlexaPi.service`
and ran AlexaPi interactively in debug mode: `/opt/AlexaPi/src/main.py -d`
which showed up this: 

> CRITICAL: Your input_device 'pulse' is invalid. Use one of the following:
> default
> null

I assume this has something to do with pulse audio, **# HOWEVER** this same message showed up for when the input device was originally plughw:CARD=Device,DEV=0 , which I later changed into pulse by doing `sudo vim /etc/opt/AlexaPi/config.yam`

Luckily, I backed up the config file before I started changing it.

I hoped that I have provided enough information for you all to help me out. I am a linux beginner user, and I hope to learn something new! :)

billcui57, 2017-08-26:
I also followed the [Running PA in system-wide mode (recommended)](https://github.com/alexa-pi/AlexaPi/wiki/Debugging#logs) help with no luck

maso27, 2017-08-26:
Have you tried replacing 'pulse' with 'default'?

billcui57, 2017-08-27:
@maso27 I replaced 'pulse' with 'default' and now it works, thanks!

closed 2017-08-27 by billcui57