alexa-pi/AlexaPi #304: CERTIFICATE_VERIFY_FAILED during setup.sh

OutsourcedGuru, 2017-08-27:
I'm guessing that the author of [this issue](https://github.com/alexa-pi/AlexaPi/issues/182) was seeing something similar but worked around the problem (in theory) by manually installing Python or one of the Python-related pieces like pip.

Following the Installation instructions today so the code should be the latest

In order to help troubleshooting, be sure to include the following information:

**Your OS (including version) where you are running AlexaPi:**
```
$ cat /etc/os-release                               
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi Zero W, 4GB microSD
```

**Python release (`python2 --version`):**
```
python2.7-dev (2.7.9-2+deb8u1)
...this is what it was trying to install during setup.sh
```

**Description of problem:**
```
From setup.sh session:
Processing triggers for systemd (215-17+deb8u7) ...
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Package 'python-pip' is not installed, so not removed
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
Searching for pip
Reading https://pypi.python.org/simple/pip/
Download error on https://pypi.python.org/simple/pip/: [Errno 104] Connection reset by peer -- Some packages may not be found!
Couldn't find index page for 'pip' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581) -- Some packages may not be found!
No local packages or download links found for pip
error: Could not find suitable distribution for Requirement.parse('pip')
```

**Expected:**
```
I expect that pypi.python.org's certificate would be in the root CA but apparently it's not.
```

**Problem-relevant `config.yaml` entries:**
```
n/a
```

**Steps to reproduce:**

1. cd /opt
2. sudo apt-get update
3. sudo apt-get upgrade
4. sudo apt-get install git
5. sudo git clone https://github.com/alexa-pi/AlexaPi.git
6. sudo ./AlexaPi/src/scripts/setup.sh

**Traceback (if applicable):**
```
n/a
```

**Additional info:**
```
$ openssl s_client -connect pypi.python.org:443
CONNECTED(00000003)
depth=2 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert High Assurance EV Root CA
verify return:1
depth=1 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert SHA2 Extended Validation Server CA
verify return:1
depth=0 businessCategory = Private Organization, 1.3.6.1.4.1.311.60.2.1.3 = US, 1.3.6.1.4.1.311.60.2.1.2 = Delaware, serialNumber = 3359300, street = 16 Allen Rd, postalCode = 03894-4801, C = US, ST = New Hampshire, L = Wolfeboro, O = Python Software Foundation, CN = www.python.org
verify return:1
---
Certificate chain
 0 s:/businessCategory=Private Organization/1.3.6.1.4.1.311.60.2.1.3=US/1.3.6.1.4.1.311.60.2.1.2=Delaware/serialNumber=3359300/street=16 Allen Rd/postalCode=03894-4801/C=US/ST=New Hampshire/L=Wolfeboro/O=Python Software Foundation/CN=www.python.org
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 Extended Validation Server CA
 1 s:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 Extended Validation Server CA
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert High Assurance EV Root CA
---
Server certificate
-----BEGIN CERTIFICATE-----
MIII9TCCB92gAwIBAgIQBfUWINUHHM+XBay/kaGttjANBgkqhkiG9w0BAQsFADB1
MQswCQYDVQQGEwJVUzEVMBMGA1UEChMMRGlnaUNlcnQgSW5jMRkwFwYDVQQLExB3
d3cuZGlnaWNlcnQuY29tMTQwMgYDVQQDEytEaWdpQ2VydCBTSEEyIEV4dGVuZGVk
IFZhbGlkYXRpb24gU2VydmVyIENBMB4XDTE3MDYxNzAwMDAwMFoXDTE4MDkyNzEy
MDAwMFowggEDMR0wGwYDVQQPDBRQcml2YXRlIE9yZ2FuaXphdGlvbjETMBEGCysG
AQQBgjc8AgEDEwJVUzEZMBcGCysGAQQBgjc8AgECEwhEZWxhd2FyZTEQMA4GA1UE
BRMHMzM1OTMwMDEUMBIGA1UECRMLMTYgQWxsZW4gUmQxEzARBgNVBBETCjAzODk0
LTQ4MDExCzAJBgNVBAYTAlVTMRYwFAYDVQQIEw1OZXcgSGFtcHNoaXJlMRIwEAYD
VQQHEwlXb2xmZWJvcm8xIzAhBgNVBAoTGlB5dGhvbiBTb2Z0d2FyZSBGb3VuZGF0
aW9uMRcwFQYDVQQDEw53d3cucHl0aG9uLm9yZzCCASIwDQYJKoZIhvcNAQEBBQAD
ggEPADCCAQoCggEBAKu1Nfs8thGqKtKu70e6jyIe8eNeD2gBbQRz8qf3MzwaK5C4
O+MeW0NMvuES4xTC1QkqDWDiZRv7TbVtfvI5URp7IjVFWulMYzDG8/MpOphIXOxC
6GGmwb/bJLYnATemEmRzK7NIDRtaBQ+jEXxIqfzDAA20Q/Ae1PIoyTbxHudXU2Pv
WyrzYeYi76SfM273SuJ4dGeTZOBEkXwVOVCEMhiIt3Fzld1TCNoYLvlKKvteKmVm
Oe9MMIuydLwkI3VI9LPFrEzhth4WEyBhgdSTog/U4eO+h0mAhtyuLSWOKUpybLIR
5YAUnFrVZtLnFxIXI3TzLnWA0BJxzIkx5cylR+kCAwEAAaOCBO8wggTrMB8GA1Ud
IwQYMBaAFD3TUKXWoK3u80pgCmXTIdT4+NYPMB0GA1UdDgQWBBSCd3T4+oG5ftt6
bybYkoATYvj/GjCCASEGA1UdEQSCARgwggEUgg53d3cucHl0aG9uLm9yZ4IPZG9j
cy5weXRob24ub3Jngg9idWdzLnB5dGhvbi5vcmeCD3dpa2kucHl0aG9uLm9yZ4IN
aGcucHl0aG9uLm9yZ4IPbWFpbC5weXRob24ub3Jngg9weXBpLnB5dGhvbi5vcmeC
FHBhY2thZ2luZy5weXRob24ub3JnghBsb2dpbi5weXRob24ub3JnghJkaXNjdXNz
LnB5dGhvbi5vcmeCDHVzLnB5Y29uLm9yZ4IHcHlwaS5pb4IMZG9jcy5weXBpLmlv
gghweXBpLm9yZ4INZG9jcy5weXBpLm9yZ4IPZG9uYXRlLnB5cGkub3JnghNkZXZn
dWlkZS5weXRob24ub3JnMA4GA1UdDwEB/wQEAwIFoDAdBgNVHSUEFjAUBggrBgEF
BQcDAQYIKwYBBQUHAwIwdQYDVR0fBG4wbDA0oDKgMIYuaHR0cDovL2NybDMuZGln
aWNlcnQuY29tL3NoYTItZXYtc2VydmVyLWcyLmNybDA0oDKgMIYuaHR0cDovL2Ny
bDQuZGlnaWNlcnQuY29tL3NoYTItZXYtc2VydmVyLWcyLmNybDBLBgNVHSAERDBC
MDcGCWCGSAGG/WwCATAqMCgGCCsGAQUFBwIBFhxodHRwczovL3d3dy5kaWdpY2Vy
dC5jb20vQ1BTMAcGBWeBDAEBMIGIBggrBgEFBQcBAQR8MHowJAYIKwYBBQUHMAGG
GGh0dHA6Ly9vY3NwLmRpZ2ljZXJ0LmNvbTBSBggrBgEFBQcwAoZGaHR0cDovL2Nh
Y2VydHMuZGlnaWNlcnQuY29tL0RpZ2lDZXJ0U0hBMkV4dGVuZGVkVmFsaWRhdGlv
blNlcnZlckNBLmNydDAMBgNVHRMBAf8EAjAAMIIB9gYKKwYBBAHWeQIEAgSCAeYE
ggHiAeAAdQCkuQmQtBhYFIe7E6LMZ3AKPDWYBPkb37jjd80OyA3cEAAAAVy24fww
AAAEAwBGMEQCIGmp6GcPeTFqyHs24BxB/Vdl5Pad4pg+P4w3Tl0qw5Q4AiBDmZzE
9W3CeQahgBE54Lo4Md8CRdGG1tgRXxBowgiD9gB2AFYUBpov18Ls0/XhvUSyPsdG
drm8mRFcwO+UmFXWidDdAAABXLbh/NYAAAQDAEcwRQIhANFDeJf6q7eriKeyUvLc
PUQo8M84UUNzje6UGR336+sqAiBNR+mU+Wh+z6SIamxQA8HdtXwKLsDB10O3WzCr
Frop5gB3AO5Lvbd1zmC64UJpH6vhnmajD35fsHLYgwDEe4l6qP3LAAABXLbh/x4A
AAQDAEgwRgIhAKu1V5iDXekw5TNvMBoZGOTkXPOZfyl7RwGB03JOzPnKAiEAlG+L
hsMLd4BuN41jTbB8dDJPJVVB3mEVC6IPY1a62lIAdgC72d+8H4pxtZOUI5eqkntH
OFeVCqtS6BqQlmQ2jh7RhQAAAVy24fzjAAAEAwBHMEUCIAhuXxQLIRomFeucS9HT
a/3jsUc8o7R7RORSiRp9Ip3TAiEAkuIgsdtSFm8qLpifl+0ngU7gxjXcAHsvuIP7
efOpu30wDQYJKoZIhvcNAQELBQADggEBAL1TGYhR3unop9X/Nu4Ym38AQ7ks992+
beK83EbLyLXjNicBR3EW86GtKJ5lBKNUq0ZNFWQjeji99Zjw1iv5GRqnAGkakolm
ZdO3kXM4nP0DEFEwEW5bLV6o+DlLMBoT0ZT/Zn6JlkCPCS2Z2J5W5rp/KNgwsMTA
Y0cdNWPYtFAEe2iJpOtIJ8uZRKCe+cj4wMuw9qEBa5nB5wgSpUB6kr5gjTDYcOCA
/wPKp3g3shPja8bkYufc2mJ79PgjZhJR6Z8jTpCY155JfLes+l1THL/L54hQ/L/M
lBvRRs8wTczjbK0lSDJIG19bfMn5Xe7UXb5aMpt6FMKIxZmsfo1ys5I=
-----END CERTIFICATE-----
subject=/businessCategory=Private Organization/1.3.6.1.4.1.311.60.2.1.3=US/1.3.6.1.4.1.311.60.2.1.2=Delaware/serialNumber=3359300/street=16 Allen Rd/postalCode=03894-4801/C=US/ST=New Hampshire/L=Wolfeboro/O=Python Software Foundation/CN=www.python.org
issuer=/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 Extended Validation Server CA
---
No client certificate CA names sent
---
SSL handshake has read 4164 bytes and written 415 bytes
---
New, TLSv1/SSLv3, Cipher is ECDHE-RSA-AES128-GCM-SHA256
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: C32242F3441303833B29F0DE70D532FEDF1FBE308A87A430B1C9D0C349999CDD
    Session-ID-ctx: 
    Master-Key: 23BE06D55E10BF9C5E2AEA0C14494414034746F78742695FB14AAAAFA18B30B8E8BCC7E8DC06CFAF556C5A7B51FB6277
    Key-Arg   : None
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 1200 (seconds)
    TLS session ticket:
    0000 - d1 cd ed 16 e2 cc ca 81-c7 6f 02 db 8e e4 b5 82   .........o......
    0010 - a0 95 b5 78 83 87 aa c1-9a 87 c3 25 7b 16 99 b9   ...x.......%{...
    0020 - 05 db ee d8 ff 3c d4 2d-fb 82 f0 0d fb c8 c6 10   .....<.-........
    0030 - 22 7e 88 eb a8 e6 d4 9a-0d eb c2 1b 7d ab 08 33   "~..........}..3
    0040 - ab 0c ea 57 c3 24 19 e5-80 bd 36 dd f9 ed df 73   ...W.$....6....s
    0050 - 1d 2d 68 33 45 6f 5e 62-ea 22 a3 18 b4 b9 49 7a   .-h3Eo^b."....Iz
    0060 - 30 02 d7 69 65 c9 c5 94-ad 34 3e b4 9f 0d fd 0e   0..ie....4>.....
    0070 - 58 47 52 45 03 b2 ef 49-db 0a 72 c2 8d e8 2e ba   XGRE...I..r.....
    0080 - e0 aa 5f 15 85 9d 44 e4-e1 08 e6 ac a5 c9 c8 3c   .._...D........<
    0090 - 20 bb 90 1d 90 d5 fb 64-89 c0 91 7f c2 bc cb 52    ......d.......R

    Start Time: 1503799571
    Timeout   : 300 (sec)
    Verify return code: 0 (ok)
---
Ctrl-C
```

OutsourcedGuru, 2017-08-28:
Should I attempt to work around the issue by manually installing something and then running the setup.sh again?

maso27, 2017-08-28:
You could try this:
`sudo apt-get install python-pip`
then re-run setup.sh.

It seems like there's been a few issues with things not getting installed via the raspbian repos or something.

OutsourcedGuru, 2017-08-28:
I'll give that a try, thanks.

Unfortunately, python.org bound their digital certificate to www.python.org not to pypi.python.org so it's invalid to use it like that.  A certificate isn't promiscuous; it's meant to be specific to a single hostname, usually.

[image]

In theory, everything would work if people used the correct host name or if python.org would split the web- and distribution-servers and bound two different certificates to them.  Just guessing, though.  It could be that their cert didn't make it into the caroot certificate.

OutsourcedGuru, 2017-08-29:
Bummer.  It looks like it's trying to uninstall PIP if it's already loaded.  And then it fails for the same reason.

```
Building dependency tree       
Reading state information... Done
The following packages were automatically installed and are no longer required:
  python-chardet python-colorama python-distlib python-html5lib
  python-ndg-httpsclient python-openssl python-pyasn1 python-requests
  python-six python-urllib3 python-wheel
Use 'apt-get autoremove' to remove them.
The following packages will be REMOVED:
  python-pip
0 upgraded, 0 newly installed, 1 to remove and 0 not upgraded.
After this operation, 481 kB disk space will be freed.
(Reading database ... 36135 files and directories currently installed.)
Removing python-pip (1.5.6-5) ...
Processing triggers for man-db (2.7.5-1~bpo8+1) ...
Searching for pip
Reading https://pypi.python.org/simple/pip/
Download error on https://pypi.python.org/simple/pip/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581) -- Some packages may not be found!
Couldn't find index page for 'pip' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581) -- Some packages may not be found!
No local packages or download links found for pip
error: Could not find suitable distribution for Requirement.parse('pip')
```

OutsourcedGuru, 2017-08-29:
I tried the version involving the time being off by more than 1000 ms, (didn't work).

```
$ cd /opt
$ sudo systemctl start ntp
$ sudo ./AlexaPi/src/scripts/setup.sh
```

I ran the following:
```
$ openssl s_client -CAfile /etc/ssl/certs/ca-certificates.crt -showcerts -connect pypi.python.org:443 < /dev/null
CONNECTED(00000003)
depth=2 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert High Assurance EV Root CA
verify return:1
depth=1 C = US, O = DigiCert Inc, OU = www.digicert.com, CN = DigiCert SHA2 Extended Validation Server CA
verify return:1
depth=0 businessCategory = Private Organization, 1.3.6.1.4.1.311.60.2.1.3 = US, 1.3.6.1.4.1.311.60.2.1.2 = Delaware, serialNumber = 3359300, street = 16 Allen Rd, postalCode = 03894-4801, C = US, ST = New Hampshire, L = Wolfeboro, O = Python Software Foundation, CN = www.python.org
verify return:1
---
Certificate chain
 0 s:/businessCategory=Private Organization/1.3.6.1.4.1.311.60.2.1.3=US/1.3.6.1.4.1.311.60.2.1.2=Delaware/serialNumber=3359300/street=16 Allen Rd/postalCode=03894-4801/C=US/ST=New Hampshire/L=Wolfeboro/O=Python Software Foundation/CN=www.python.org
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 Extended Validation Server CA
-----BEGIN CERTIFICATE-----
---snip---
-----END CERTIFICATE-----
 1 s:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 Extended Validation Server CA
   i:/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert High Assurance EV Root CA
-----BEGIN CERTIFICATE-----
---snip---
-----END CERTIFICATE-----
---
Server certificate
subject=/businessCategory=Private Organization/1.3.6.1.4.1.311.60.2.1.3=US/1.3.6.1.4.1.311.60.2.1.2=Delaware/serialNumber=3359300/street=16 Allen Rd/postalCode=03894-4801/C=US/ST=New Hampshire/L=Wolfeboro/O=Python Software Foundation/CN=www.python.org
issuer=/C=US/O=DigiCert Inc/OU=www.digicert.com/CN=DigiCert SHA2 Extended Validation Server CA
---
No client certificate CA names sent
---
SSL handshake has read 4164 bytes and written 415 bytes
---
New, TLSv1/SSLv3, Cipher is ECDHE-RSA-AES128-GCM-SHA256
Server public key is 2048 bit
Secure Renegotiation IS supported
Compression: NONE
Expansion: NONE
SSL-Session:
    Protocol  : TLSv1.2
    Cipher    : ECDHE-RSA-AES128-GCM-SHA256
    Session-ID: D8AAE0CF71D9E9C65C7C106DAE6A4FE73C72287B89AD07CC43468F32632ACB62
    Session-ID-ctx: 
    Master-Key: AD3F3C5D68B835D75ACEB11A16E711299B3442D1187A41D66DE4E8C06738CB542591FC2944398DA635E98EC794D449E0
    Key-Arg   : None
    PSK identity: None
    PSK identity hint: None
    SRP username: None
    TLS session ticket lifetime hint: 1200 (seconds)
    TLS session ticket:
    ---snip---

    Start Time: 1503973768
    Timeout   : 300 (sec)
    Verify return code: 0 (ok)
---
DONE
```

It's expected to return 0 (OK) from that "verify return" line.  It fails also with www.python.org, for what it's worth.

```
$ sudo update-ca-certificates
$ sudo ./AlexaPi/src/scripts/setup.sh
```

Still the same result, unfortunately.

maso27, 2017-08-29:
So it looks like the uninstallation of pip is on purpose. It's being reinstalled via a different method.
This is in the script:
```
apt-get -y remove python-pip
run_python -m easy_install pip
```
(Which translates into `python2 -m easy_install pip`)

I'm not sure why the different method is employed exactly.  
But it might help point you towards the solution you need. Maybe clean reinstalling python2 or something?

Alternatively, you could try editing _/opt/AlexaPi/src/scripts/inc/os/debian.sh_ to read
`apt-get -y install python-pip`
(_without_ the easy_install step)

...and see what happens.

maso27, 2017-08-29:
Quick question: is the date and time correct on your rPi?

OutsourcedGuru, 2017-08-29:
Yes, it is.

Noting that the Raspberry Pi Zero W is detected as Debian:
```
$ sudo nano /opt/AlexaPi/src/scripts/inc/os/debian.sh

function install_os {
    apt-get update
    apt-get install curl git build-essential python-dev python-setuptools swig $
    #apt-get -y remove python-pip
    #run_python -m easy_install pip
}
# Commented out both the remove/reinstall lines for pip

$ sudo ./AlexaPi/src/scripts/setup.sh
...
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
Downloading/unpacking requests>=2.13.0 (from -r ./requirements.txt (line 1))
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1178, in prepare_files
    url = finder.find_requirement(req_to_install, upgrade=self.upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 194, in find_requirement
    page = self._get_page(main_index_url, req)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_page
    session=self.session,
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 694, in get_page
    req, link, "connection error: %s" % exc, url,
TypeError: __str__ returned non-string (type Error)

Storing debug log for failure in /root/.pip/pip.log
```

OutsourcedGuru, 2017-08-29:
```
$ sudo python -m easy_install pip
Searching for pip
Best match: pip 1.5.6
Adding pip 1.5.6 to easy-install.pth file
Installing pip script to /usr/local/bin
Installing pip2.7 script to /usr/local/bin
Installing pip2 script to /usr/local/bin

Using /usr/lib/python2.7/dist-packages
Processing dependencies for pip
Finished processing dependencies for pip

$ sudo ./AlexaPi/src/scripts/setup.sh
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
Downloading/unpacking requests>=2.13.0 (from -r ./requirements.txt (line 1))
Cleaning up...
Exception:
Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/usr/lib/python2.7/dist-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/usr/lib/python2.7/dist-packages/pip/req.py", line 1178, in prepare_files
    url = finder.find_requirement(req_to_install, upgrade=self.upgrade)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 194, in find_requirement
    page = self._get_page(main_index_url, req)
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 568, in _get_page
    session=self.session,
  File "/usr/lib/python2.7/dist-packages/pip/index.py", line 694, in get_page
    req, link, "connection error: %s" % exc, url,
TypeError: __str__ returned non-string (type Error)

Storing debug log for failure in /root/.pip/pip.log
```

OutsourcedGuru, 2017-08-29:
^ So it's failing on the requirements.txt file's first line for v2.13.0 of requests.  Sounds like pip isn't happy so this attempt at manually installing it outside of the system isn't working.

maso27, 2017-08-29:
We'll, that's probably why it's been changed.

Meanwhile, why isn't easy_install working?
I'm sorry to say that I don't know.

OutsourcedGuru, 2017-08-29:
Easy_install doesn't seem to be happy under the control of setup.sh, somehow.  I've reverted the debian.sh and am running it again.

OutsourcedGuru, 2017-08-29:
Removing python-pip (1.5.6-5) ...
Processing triggers for man-db (2.7.5-1~bpo8+1) ...
Searching for pip
Reading https://pypi.python.org/simple/pip/
Download error on https://pypi.python.org/simple/pip/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581) -- Some packages may not be found!
Couldn't find index page for 'pip' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading https://pypi.python.org/simple/
Download error on https://pypi.python.org/simple/: [SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:581) -- Some packages may not be found!
No local packages or download links found for pip
error: Could not find suitable distribution for Requirement.parse('pip')

OutsourcedGuru, 2017-08-29:
$ sudo nano /usr/lib/python2.7/dist-packages/setuptools/ssl_support.py
# add "ssl = None" after the try section near the top to override ssl completely
# Per https://stackoverflow.com/questions/32894267/how-to-get-easy-install-to-ignore-certifcate

... and it appears to be installing now.  We'll see...

maso27, 2017-08-29:
Fantastic!
I was about to suggest manually pointing at one of the tar.gz files at https://pypi.python.org/simple, but I think this is a better workaround.

OutsourcedGuru, 2017-08-29:
Setup.sh is chugging right along.

1. manually install python
2. tweak the ssl_support.py file as suggested to turn off ssl
3. run your script
etc

Sounds like a workable solution to me.  You can always toggle it back on after the install.

closed 2017-08-29 by OutsourcedGuru