alexa-pi/AlexaPi #306: AlexaPi failed on startup (Read the debug section, no help)

billcui57, 2017-08-28:
This is weird...
On startup a line flashed by reading "Starting Alexapi for your devices" 
However when I typed in `sudo systemctl status AlexaPi.service`
This appeared: 

>   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Mon 2017-08-28 06:27:47 UTC; 16s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 454 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 454 (code=exited, status=1/FAILURE)

Aug 28 06:27:47 raspberrypi python[454]:     self.current_event = self.state()
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
Aug 28 06:27:47 raspberrypi python[454]:     "expected <block end>, but found %r" % token.id, token.start_mark)
Aug 28 06:27:47 raspberrypi python[454]: yaml.parser.ParserError: while parsing a block mapping
Aug 28 06:27:47 raspberrypi python[454]:   in "/etc/opt/AlexaPi/config.yaml", line 7, column 1
Aug 28 06:27:47 raspberrypi python[454]: expected <block end>, but found '<block mapping start>'
Aug 28 06:27:47 raspberrypi python[454]:   in "/etc/opt/AlexaPi/config.yaml", line 14, column 3
Aug 28 06:27:47 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Aug 28 06:27:47 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Aug 28 06:27:47 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
~
I dont see any red stuff so I think nothing is wrong. I have no idea. 

I stopped the service and then started it again.
Checked status: 

> ● AlexaPi.service - Alexa client for all your devices
>    Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
>    Active: failed (Result: exit-code) since Mon 2017-08-28 06:33:01 UTC; 416ms ago
>      Docs: https://github.com/alexa-pi/AlexaPi/wiki
>   Process: 777 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
>  Main PID: 777 (code=exited, status=1/FAILURE)
> 
> Aug 28 06:33:01 raspberrypi python[777]:     self.current_event = self.state()
> Aug 28 06:33:01 raspberrypi python[777]:   File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
> Aug 28 06:33:01 raspberrypi python[777]:     "expected <block end>, but found %r" % token.id, token.start_mark)
> Aug 28 06:33:01 raspberrypi python[777]: yaml.parser.ParserError: while parsing a block mapping
> Aug 28 06:33:01 raspberrypi python[777]:   in "/etc/opt/AlexaPi/config.yaml", line 7, column 1
> Aug 28 06:33:01 raspberrypi python[777]: expected <block end>, but found '<block mapping start>'
> Aug 28 06:33:01 raspberrypi python[777]:   in "/etc/opt/AlexaPi/config.yaml", line 14, column 3
> Aug 28 06:33:01 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
> Aug 28 06:33:01 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
> Aug 28 06:33:01 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
> 
> 
Stopped and started it again:

>  AlexaPi.service - Alexa client for all your devices
>    Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
>    Active: active (running) since Mon 2017-08-28 06:34:04 UTC; 2s ago
>      Docs: https://github.com/alexa-pi/AlexaPi/wiki
>  Main PID: 811 (python)
>    CGroup: /system.slice/AlexaPi.service
>            └─811 /usr/bin/python /opt/AlexaPi/src/main.py --daemon
> 
> Aug 28 06:34:04 raspberrypi systemd[1]: Started Alexa client for all your devices.
> 
?!?!?!?!?!?!?!?! how come it worked this time????

Apart from that issue, I never hear Alexa say "Hello" 
Possibly something wrong with my mic or speaker? But when i tested them they worked flawlessly
Thank you!
`

billcui57, 2017-08-28:
Here is the log from sudo journalctl -u AlexaPi.service

Aug 28 06:27:41 raspberrypi systemd[1]: Started Alexa client for all your devices.
Aug 28 06:27:47 raspberrypi python[454]: Traceback (most recent call last):
Aug 28 06:27:47 raspberrypi python[454]:   File "/opt/AlexaPi/src/main.py", line 71, in <module>
Aug 28 06:27:47 raspberrypi python[454]:     config = yaml.load(stream)
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
Aug 28 06:27:47 raspberrypi python[454]:     return loader.get_single_data()
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
Aug 28 06:27:47 raspberrypi python[454]:     node = self.get_single_node()
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
Aug 28 06:27:47 raspberrypi python[454]:     document = self.compose_document()
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
Aug 28 06:27:47 raspberrypi python[454]:     node = self.compose_node(None, None)
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
Aug 28 06:27:47 raspberrypi python[454]:     node = self.compose_mapping_node(anchor)
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
Aug 28 06:27:47 raspberrypi python[454]:     while not self.check_event(MappingEndEvent):
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
Aug 28 06:27:47 raspberrypi python[454]:     self.current_event = self.state()
Aug 28 06:27:47 raspberrypi python[454]:   File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
Aug 28 06:27:47 raspberrypi python[454]:     "expected <block end>, but found %r" % token.id, token.start_mark)
Aug 28 06:27:47 raspberrypi python[454]: yaml.parser.ParserError: while parsing a block mapping
Aug 28 06:27:47 raspberrypi python[454]:   in "/etc/opt/AlexaPi/config.yaml", line 7, column 1
Aug 28 06:27:47 raspberrypi python[454]: expected <block end>, but found '<block mapping start>'
Aug 28 06:27:47 raspberrypi python[454]:   in "/etc/opt/AlexaPi/config.yaml", line 14, column 3
Aug 28 06:27:47 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Aug 28 06:27:47 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Aug 28 06:27:47 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.
Aug 28 06:32:59 raspberrypi systemd[1]: Started Alexa client for all your devices.
Aug 28 06:33:01 raspberrypi python[777]: Traceback (most recent call last):
Aug 28 06:33:01 raspberrypi python[777]:   File "/opt/AlexaPi/src/main.py", line 71, in <module>
Aug 28 06:33:01 raspberrypi python[777]:     config = yaml.load(stream)
Aug 28 06:33:01 raspberrypi python[777]:   File "/usr/local/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
Aug 28 06:33:01 raspberrypi python[777]:     return loader.get_single_data()
Aug 28 06:33:01 raspberrypi python[777]:   File "/usr/local/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
Aug 28 06:33:01 raspberrypi python[777]:     node = self.get_single_node()
Aug 28 06:33:01 raspberrypi python[777]:   File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
Aug 28 06:33:01 raspberrypi python[777]:     document = self.compose_document()
Aug 28 06:33:01 raspberrypi python[777]:   File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
Aug 28 06:33:01 raspberrypi python[777]:     node = self.compose_node(None, None)
Aug 28 06:33:01 raspberrypi python[777]:   File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
Aug 28 06:33:01 raspberrypi python[777]:     node = self.compose_mapping_node(anchor)

billcui57, 2017-08-28:
From sudo systemctl stop AlexaPi.service:

Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 71, in <module>
    config = yaml.load(stream)
  File "/usr/local/lib/python2.7/dist-packages/yaml/__init__.py", line 71, in load
    return loader.get_single_data()
  File "/usr/local/lib/python2.7/dist-packages/yaml/constructor.py", line 37, in get_single_data
    node = self.get_single_node()
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
  File "/usr/local/lib/python2.7/dist-packages/yaml/composer.py", line 127, in compose_mapping_node
    while not self.check_event(MappingEndEvent):
  File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
  File "/usr/local/lib/python2.7/dist-packages/yaml/parser.py", line 439, in parse_block_mapping_key
    "expected <block end>, but found %r" % token.id, token.start_mark)
yaml.parser.ParserError: while parsing a block mapping
  in "/etc/opt/AlexaPi/config.yaml", line 7, column 1
expected <block end>, but found '<block mapping start>'
  in "/etc/opt/AlexaPi/config.yaml", line 14, column 3

billcui57, 2017-08-28:
`aplay /usr/share/sounds/alsa/Front_Center.wav`
and
```
arecord -f S16_LE -r 16000 -D sysdefault:CARD=Device test.wav 
aplay -f S16_LE -r 16000 test.wav 

```
worked fine, but I am unable to check AlexaPi's recordings.
I ran in debug mode after I stopped the service, but I can't find the AlexaPi-runtime-xxx directory in /tmp

billcui57, 2017-08-28:
xxx for the string of course

maso27, 2017-08-28:
Not sure what all has happened here, but have you looked at line 7, and then line 14, in _/etc/opt/AlexaPi/config.yaml_?

That's where it's telling you the failure is.

billcui57, 2017-08-29:
Line 7 is the debug or info option
Line 14 is the allow unlisted audio device option
I am not sure what i am looking for to fix this

maso27, 2017-08-29:
It's saying there's a quotation mark where it doesn't expect one on line 7.
In the template, line 7 is `logging: INFO`
Have you changed this?

renekliment, 2017-10-08:
Looks like syntax error in the YAML config.
Closing due to inactivity. Feel free to reopen.

closed 2017-10-08 by renekliment