alexa-pi/AlexaPi #307: Setup.sh prompts should match Alexa Voice attribute names

OutsourcedGuru, 2017-08-29:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Raspian Lite Jessie 8
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi Zero W
```

**Python release (`python2 --version`):**
```
2.7.9
```

**Description of problem:**
```
Creating configuration file /etc/opt/AlexaPi/config.yaml ...
Enter your Device Type ID []: <- This should ask for "PRODUCT ID" instead
Enter your Security Profile Description []: <- Okay as is
Enter your Security Profile ID []: <- Okay as is
Enter your Client ID []: <- Yes, but it's hidden on the Security Profile page, perhaps "Security Profile -> Client ID"
Enter your Client Secret []: <- For the same reason, perhaps "Security Profile -> Client Secret"
```

**Expected:**
I had to review the script to see what device ID actually was.

dikkedimi, 2017-10-04:
Yeah, this would save a lot of people a lot of confusion and frustration. Of course, forking, modifying and then merging would be an option, this is Github after all ;)

dikkedimi, 2017-10-04:
confirmed my changes working, except for an error which I also got with the original package.

`ALSAAudioError: Device or resource busy [dsnoop:CARD=C920,DEV=0]`

Still looking for a way to solve that, but I had zero problems authenticating with Alexa services in any case, which was my previous obstacle. Guess this change was overdue.

OutsourcedGuru, 2017-10-05:
I'm still fussing with mine without a full unit test successful. Until I can confirm that, I'd be shy on the fork + edit + pull request.

dikkedimi, 2017-10-05:
I got an error from the web_auth.py script before, seems to have solved that as well.