alexa-pi/AlexaPi #308: URLError: [Errno 111]

Little-John21, 2017-08-29:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```Linux raspberrypi 4.9.45-v7+ #1031 SMP Fri Aug 25 19:02:16 BST 2017 armv7l GNU/Linux
```

**Your hardware platform and model you are running on:**
```
? raspberryPi 3
```

**Python release (`python2 --version`):**
```
Python 2.7.13
```

**Description of problem:**
```
AlexaPi not starting on reboot or as manual. Gives Errno 111 error.
```

**Expected:**
```
it to work
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. run program
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
`pi@raspberrypi:~ $ sudo systemctl start AlexaPi.service
pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor pres
   Active: failed (Result: exit-code) since Tue 2017-08-29 10:34:57 EDT; 4s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1440 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (cod
 Main PID: 1440 (code=exited, status=1/FAILURE)

Aug 29 10:34:57 raspberrypi python[1440]: surround51:CARD=Device,DEV=0
Aug 29 10:34:57 raspberrypi python[1440]: surround71:CARD=Device,DEV=0
Aug 29 10:34:57 raspberrypi python[1440]: iec958:CARD=Device,DEV=0
Aug 29 10:34:57 raspberrypi python[1440]: dmix:CARD=Device,DEV=0
Aug 29 10:34:57 raspberrypi python[1440]: dsnoop:CARD=Device,DEV=0
Aug 29 10:34:57 raspberrypi python[1440]: hw:CARD=Device,DEV=0
Aug 29 10:34:57 raspberrypi python[1440]: plughw:CARD=Device,DEV=0
Aug 29 10:34:57 raspberrypi systemd[1]: AlexaPi.service: Main process exited, co
Aug 29 10:34:57 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed sta
[... 236 lines cut ...]
2017-08-29 10:36:32 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-08-29 10:36:32 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 10:36:32 DEBUG: Indicating Success
2017-08-29 10:36:32 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=success
2017-08-29 10:36:32 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 10:36:34 DEBUG: Sending MM Heatbeat
2017-08-29 10:36:34 ERROR: URLError: [Errno 111] Connection refused
INFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 29.01 -15.74  7.91 -6.91  4.85  1.43 -9.23  3.60  2.83  3.69  1.54  7.85  6.88 >
2017-08-29 10:36:39 DEBUG: Sending MM Heatbeat
2017-08-29 10:36:39 ERROR: URLError: [Errno 111] Connection refused
INFO: cmn_live.c(88): Update from < 29.01 -15.74  7.91 -6.91  4.85  1.43 -9.23  3.60  2.83  3.69  1.54  7.85  6.88 >
INFO: cmn_live.c(105): Update to   < 28.29 -12.64  3.79 -6.92  8.02 -2.04 -10.39  6.68  5.54  1.86  0.81  6.97  6.97 >
^C2017-08-29 10:36:44 DEBUG: Cleaning up Magic Mirror platform
2017-08-29 10:36:44 DEBUG: Sending MM Heatbeat
2017-08-29 10:36:44 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 10:36:44 DEBUG: Cleaning up playback handler
INFO: cmn_live.c(88): Update from < 28.29 -12.64  3.79 -6.92  8.02 -2.04 -10.39  6.68  5.54  1.86  0.81  6.97  6.97 >
INFO: cmn_live.c(105): Update to   < 25.87 -10.35  3.97 -5.07  5.98  1.06 -8.60  5.23  5.37  2.47  1.69  5.72  4.34 >
^C2017-08-29 10:36:49 DEBUG: Cleaning up Magic Mirror platform
2017-08-29 10:36:49 DEBUG: Cleaning up playback handler
Exception OSError: OSError(2, 'No such file or directory') in <module 'threading' from '/usr/lib/python2.7/threading.pyc'> ignored
pi@raspberrypi:~ $ 
```
im not smart at all this at all took me a few days just to install and run the magic mirror on my Pi. if anyone has any incite it would be appreciated.

maso27, 2017-08-29:
Your report is truncated, so it's hard to read the errors, but this line:

`Aug 29 10:34:52 raspberrypi python[1424]: CRITICAL: Your input_device 'default' is invalid. Use one`

...is saying that "default" is set as your _input_device_.  

Further on that line it will make suggestions for what to replace it with, based on your particular audio setup.
Try changing the _input_device_ in _/etc/opt/AlexaPi/config.yaml_ to something else on that list.

(I suspect you'll want to use `plughw:CARD=Device,DEV=0`).

Little-John21, 2017-08-29:
Thanks Mason ill give it a try i have been trying for a few days to get my alsa to save when i change to my USB sound card but it never retains it. i will add to the yaml file now.

Little-John21, 2017-08-29:
ok fixed that although when running logs it still says it but in the yaml file its saved and reading fine.

also i am getting further when running the debug. it now shows 

`pi@raspberrypi:~ $ sudo systemctl stop AlexaPi.service
pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py -d

2017-08-29 11:54:05 DEBUG: Initialising Magic Mirror platorm
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-10
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
[... 125 lines cut ...]
2017-08-29 11:54:09 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=playback
2017-08-29 11:54:09 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 11:54:11 DEBUG: Player State: State.Ended
2017-08-29 11:54:11 DEBUG: Finished play. 
2017-08-29 11:54:11 DEBUG: Indicate Stop Playing
2017-08-29 11:54:11 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=idle
2017-08-29 11:54:11 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 11:54:11 DEBUG: Indicating Success
2017-08-29 11:54:11 DEBUG: Calling URL: http://localhost:8080/alexapi?action=AVSSTATUS&status=success
2017-08-29 11:54:11 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 11:54:14 DEBUG: Sending MM Heatbeat
2017-08-29 11:54:14 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 11:54:19 DEBUG: Sending MM Heatbeat
2017-08-29 11:54:19 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 11:54:24 DEBUG: Sending MM Heatbeat
2017-08-29 11:54:24 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 11:54:29 DEBUG: Sending MM Heatbeat
2017-08-29 11:54:29 ERROR: URLError: [Errno 111] Connection refused
^C2017-08-29 11:54:34 DEBUG: Cleaning up Magic Mirror platform
2017-08-29 11:54:34 DEBUG: Sending MM Heatbeat
2017-08-29 11:54:34 ERROR: URLError: [Errno 111] Connection refused
2017-08-29 11:54:34 DEBUG: Cleaning up playback handler
^C2017-08-29 11:54:39 DEBUG: Cleaning up Magic Mirror platform
2017-08-29 11:54:39 DEBUG: Cleaning up playback handler
`

maso27, 2017-08-29:
I have no experience with Magic Mirror.  You may have to do some independent debug on that.

But have you somewhere configured an audio device as "USB Audio Device"?  It looks like there's an error there.

You can use `aplay -L` to list valid playback devices, and put the appropriate one under _output_device_ in _config.yaml_.

Usually the best option to use is one that starts with "plughw" instead of "hw".

Little-John21, 2017-08-29:
when Magic mirror is open i now get

`pi@raspberrypi:~ $ /opt/AlexaPi/src/main.py -d
2017-08-29 12:44:07 DEBUG: Initialising Magic Mirror platorm
INFO: pocketsphinx.c(152): Parsed model-specific feature parameters from /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/feat.params
Current configuration:
[NAME]			[DEFLT]		[VALUE]
-agc			none		none
-agcthresh		2.0		2.000000e+00
-allphone				
-allphone_ci		no		no
-alpha			0.97		9.700000e-01
-ascale			20.0		2.000000e+01
-aw			1		1
-backtrace		no		no
-beam			1e-48		1.000000e-48
-bestpath		yes		yes
-bestpathlw		9.5		9.500000e+00
-ceplen			13		13
-cmn			live		batch
-cmninit		40,3,-1		41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,-1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen		no		no
-debug					0
-dict					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
-dictcase		no		no
-dither			no		no
-doublebw		no		no
-ds			1		1
-fdict					
-feat			1s_c_d_dd	1s_c_d_dd
-featparams				
-fillprob		1e-8		1.000000e-08
-frate			100		100
-fsg					
-fsgusealtpron		yes		yes
-fsgusefiller		yes		yes
-fwdflat		yes		yes
-fwdflatbeam		1e-64		1.000000e-64
-fwdflatefwid		4		4
-fwdflatlw		8.5		8.500000e+00
-fwdflatsfwin		25		25
-fwdflatwbeam		7e-29		7.000000e-29
-fwdtree		yes		yes
-hmm					/usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us
-input_endian		little		little
-jsgf					
-keyphrase				alexa
-kws					
-kws_delay		10		10
-kws_plp		1e-1		1.000000e-01
-kws_threshold		1		1.000000e-10
-latsize		5000		5000
-lda					
-ldadim			0		0
-lifter			0		22
-lm					
-lmctl					
-lmname					
-logbase		1.0001		1.000100e+00
-logfn					
-logspec		no		no
-lowerf			133.33334	1.300000e+02
-lpbeam			1e-40		1.000000e-40
-lponlybeam		7e-29		7.000000e-29
-lw			6.5		6.500000e+00
-maxhmmpf		30000		30000
-maxwpf			-1		-1
-mdef					
-mean					
-mfclogdir				
[... 130 lines cut ...]
2017-08-29 12:44:13 ERROR: URLError: Not Found
2017-08-29 12:44:16 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:16 ERROR: URLError: Not Found
2017-08-29 12:44:21 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:21 ERROR: URLError: Not Found
2017-08-29 12:44:26 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:26 ERROR: URLError: Not Found
2017-08-29 12:44:31 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:31 ERROR: URLError: Not Found
2017-08-29 12:44:36 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:36 ERROR: URLError: Not Found
2017-08-29 12:44:41 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:41 ERROR: URLError: Not Found
2017-08-29 12:44:46 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:46 ERROR: URLError: Not Found
2017-08-29 12:44:51 DEBUG: Sending MM Heatbeat
2017-08-29 12:44:51 ERROR: URLError: Not Found
^C2017-08-29 12:44:53 DEBUG: Cleaning up Magic Mirror platform
2017-08-29 12:44:54 DEBUG: Cleaning up playback handler
INFO: kws_search.c(448): TOTAL kws 0.00 CPU nan xRT
INFO: kws_search.c(451): TOTAL kws 0.00 wall nan xRT
pi@raspberrypi:~ $ 
`

thanks ive set it up in the config.yaml file

renekliment, 2017-10-08:
Again, misconfigured MagicMirror and audio ...

closed 2017-10-08 by renekliment