alexa-pi/AlexaPi #309: Crashes when asking for weather

throwaway789, 2017-08-29:
In order to help troubleshooting, be sure to include the following information:

**Your OS (including version) where you are running AlexaPi:**
```
Rasbian stretch
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 3B
```

**Python release (`python2 --version`):**
```
Python 2.7.13
```

**Description of problem:**
```
There is a Traceback when asking for the weather or the temperature. Other queries seem to work fine. Location has been set via alexa.amazon.com. That site also shows that Amazon is receiving the query for the weather.
```

**Expected:**
```
Not a crash
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. "Alexa, what is the weather"
2. "Alexa, what is the temperature"
3. 

**Traceback (if applicable):**
```
Aug 29 08:36:31 alexa1 python[1017]: Traceback (most recent call last):
Aug 29 08:36:31 alexa1 python[1017]:   File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
Aug 29 08:36:31 alexa1 python[1017]:     self.run()
Aug 29 08:36:31 alexa1 python[1017]:   File "/usr/lib/python2.7/threading.py", line 754, in run
Aug 29 08:36:31 alexa1 python[1017]:     self.__target(*self.__args, **self.__kwargs)
Aug 29 08:36:31 alexa1 python[1017]:   File "/opt/AlexaPi/src/main.py", line 542, in trigger_process
Aug 29 08:36:31 alexa1 python[1017]:     alexa_speech_recognizer(audio_stream)
Aug 29 08:36:31 alexa1 python[1017]:   File "/opt/AlexaPi/src/main.py", line 351, in alexa_speech_recognizer
Aug 29 08:36:31 alexa1 python[1017]:     process_response(resp)
Aug 29 08:36:31 alexa1 python[1017]:   File "/opt/AlexaPi/src/main.py", line 442, in process_response
Aug 29 08:36:31 alexa1 python[1017]:     with open(filename, 'wb') as f:
Aug 29 08:36:31 alexa1 python[1017]: IOError: [Errno 2] No such file or directory: '/tmp/AlexaPi-runtime-wQt9e2/4855a0d9-51b0-4879-a
```

**Additional info:**
```

```

chileno01, 2017-08-31:
Hello:
I had the same problem. The reason is that the application (main.py) creates temporary filenames using the names sent by Amazon. In the case of weather answers this may include an slash "/" causing the creation of the temporary file to fail since the "/" is interpreted as a directory.
I was able to solve it adding a couple of lines to main.py in order to replace the "/" character with something safer. I can post it here if needed. Let me know.

Daraviace, 2017-08-31:
@chileno01 Please could you post it here,am having the same problem. Thanks

chileno01, 2017-08-31:
I was about to, but the workaround didn't work longer as another issue arised with the "#" character.
The filename created to write the file is based in the 'Content-ID' key in the payload but the filename used to read the file later for playing the file is based in the 'audioContent' key. This last one replaces "#" with ".hashMark." causing the mismatch.
I'll try to solve it but I think that the real solution is to create the filename without using the contents of the answer.

throwaway789, 2017-08-31:
Thanks chileno01, that was the issue. Here's a hacky fix. Replace line 441:
`filename = tmp_path + payload.get('Content-ID').strip("<>") + ".mp3"`
with:
`mp3_file = payload.get('Content-ID').strip("<>") + ".mp3"`
`filename = tmp_path + mp3_file.replace('/','-')`

and replace line 455:
`player.play_speech(mrl_fix("file://" + tmp_path + directive['payload']['audioContent'].lstrip("cid:") + ".mp3"))`
with:
`mp3_file = directive['payload']['audioContent'].lstrip("cid:") + ".mp3"`
`mp3_file = mp3_file.replace('/', '-')`
`player.play_speech(mrl_fix("file://" + tmp_path + mp3_file))`

chileno01, 2017-09-01:
Thanks @throwaway789 , that's a good solution. However, debugging this problem I found that there is still a chance that the program will crash if the file name starts with c, i or d. This is because the usage of lstrip.
Consider for example this code:
`a="cid:cjsjjsjjjs992"
print a
a=a.lstrip("cid:")
print a
`
It should print "cjsjjsjjjs992" but actually prints "jsjjsjjjs992"
After replacing lstrip("cid:") with replace("cid:","",1) it haven't found any more issues.
Regards

throwaway789, 2017-09-01:
Good catch. There is another instance of `lstrip("cid")` in the `play_playlist` function that should be replaced similarly.

chileno01, 2017-09-04:
Hello @stevegolf9. You need to change main.py which is usually in /opt/AlexaPi/src

stevegolf9, 2017-09-04:
> Preview: Hello @stevegolf9. You need to change main.py which is usuall
> This email is forwarded from a MASKED EMAIL you created using Blur
> <https://dnt.abine.com/#help/faq/faq-whataremaskedemails>.
> IF THIS IS SPAM, CLICK HERE TO BLOCK.
> <https://dnt.abine.com/#/block_email/4d498e45@opayq.com/
>
> Want to shop safely and privately online? Get Blur Premium
> <https://dnt.abine.com/#premium>.
>
> Hello @stevegolf9 <https://github.com/stevegolf9>. You need to change
> main.py which is usually in /opt/AlexaPi/src

Thanks. I did figure it out eventually and got it all working. Thanks in 
any case for getting back to me. I appreciate it very much.

steve

Tanonic, 2017-09-04:
I tried the hack as mentioned here but it's not working, still getting the same error message.

stevegolf9, 2017-09-05:
> Preview: I tried the hack as mentioned here but it's not working, stil

I edited the file per the instructions and it worked like a charm ( the 
second  time anyways...). I reinstalled Alexa, reedited the file and it 
worked....

chileno01, 2017-09-05:
Backup your /opt/AlexaPi/src/main.py and try the attached one after unzipping it
Let us know if it worked. Maybe you've hit another strange case.

[main.zip](https://github.com/alexa-pi/AlexaPi/files/1275871/main.zip)

Tanonic, 2017-09-05:
@chileno01 still having the problem even with your main.py

chileno01, 2017-09-05:
Could you please run main.py -d and post the log with the problem?

stevegolf9, 2017-09-05:
> Preview: Could you please run main.py -d and post the log with the pro

Its all working ok now. the copy of your main.py was helpful since I 
wasnt sure what the second sections edits were. Your file cleared that 
up for me. Thanks for all your help.

steve

Tanonic, 2017-09-07:
@chileno01 I installed AlexaPi on a desktop now and there your main.py works without a problem.

chileno01, 2017-09-07:
@Tanonic ...good news. It's still strange because mine is working in the Raspberry Pi without problems. Let me know if you are still having trouble in the Pi

Tanonic, 2017-09-15:
@chileno01 I think I'm just gonna let it be on the pi and use alexa on the desktop it works on

ViennaMike, 2017-09-15:
@chileno01: This worked on my Raspberry Pi. By the way, there's a similar but different problem when using Sox and Amazon returns mp3's with expiration date parameters after the URL caused Sox to fail with an invalid filename error.

jaimeburnap, 2017-09-16:
I did the fix that @throwaway789 mentioned, but then was getting filename too long errors. I replace the .replace() with hashlib.md5() and it's been working great.

rjoo, 2017-09-17:
@jaimeburnap can you clarify where and how you utilized `hashlib.md5()` to fix the filename too long errors?

jaimeburnap, 2017-09-17:
filename = tmp_path + hashlib.md5(payload.get('Content-ID').strip("<>")).hexdigest() + ".mp3"

player.play_speech(mrl_fix("file://" + tmp_path + hashlib.md5(directive['payload']['audioContent'].lstrip("cid:")).hexdigest() + ".mp3"))

mariuszkobak, 2017-09-19:
@jaimeburnap 

after implementing your change I got this. Can you help?

wrz 19 21:35:34 raspberrypi python[607]: self.run()
wrz 19 21:35:34 raspberrypi python[607]: File "/usr/lib/python2.7/threading.py", line 763, in run
wrz 19 21:35:34 raspberrypi python[607]: self.__target(*self.__args, **self.__kwargs)
wrz 19 21:35:34 raspberrypi python[607]: File "/opt/AlexaPi/src/main.py", line 543, in trigger_process
wrz 19 21:35:34 raspberrypi python[607]: alexa_speech_recognizer(audio_stream)
wrz 19 21:35:34 raspberrypi python[607]: File "/opt/AlexaPi/src/main.py", line 352, in alexa_speech_recognizer
wrz 19 21:35:34 raspberrypi python[607]: process_response(resp)
wrz 19 21:35:34 raspberrypi python[607]: File "/opt/AlexaPi/src/main.py", line 442, in process_response
wrz 19 21:35:34 raspberrypi python[607]: filename = tmp_path + hashlib.md5(payload.get('Content-ID').strip("<>")).hexdigest() + ".mp3"
wrz 19 21:35:34 raspberrypi python[607]: NameError: global name 'hashlib' is not defined

ViennaMike, 2017-09-19:
@mariuszkobak Did you remember to import hashlib at the start of the module?

mariuszkobak, 2017-09-19:
@ViennaMike how to do this? :)

mariuszkobak, 2017-09-20:
You were right about importing hashlib. 
Right now I have below issue. Do you know what can be wrong?

wrz 20 17:56:08 raspberrypi python[612]: [73253f20] filesystem access error: cannot open file /tmp/AlexaPi-runtime-9NnCif/5a6c25ab013f8e7b2e907b4f8b34f3ce.mp3 (No such file or directory)
wrz 20 17:56:08 raspberrypi python[612]: [73253f20] core access error: File reading failed
wrz 20 17:56:08 raspberrypi python[612]: [73253f20] core access error: VLC could not open the file "/tmp/AlexaPi-runtime-9NnCif/5a6c25ab013f8e7b2e907b4f8b34f3ce.mp3" (Nie ma takiego pliku ani katalogu).
wrz 20 17:56:08 raspberrypi python[612]: [6ee02eb8] core input error: open of `file:///tmp/AlexaPi-runtime-9NnCif/5a6c25ab013f8e7b2e907b4f8b34f3ce.mp3' failed
wrz 20 17:56:08 raspberrypi python[612]: [6ee02eb8] core input error: Your input can't be opened
wrz 20 17:56:08 raspberrypi python[612]: [6ee02eb8] core input error: VLC is unable to open the MRL 'file:///tmp/AlexaPi-runtime-9NnCif/5a6c25ab013f8e7b2e907b4f8b34f3ce.mp3'. Check the log for details.

renekliment, 2017-09-20:
If you guys can make a PR, I'll be happy to merge it.

jaimeburnap, 2017-09-20:
@renekliment done
https://github.com/alexa-pi/AlexaPi/pull/314

chileno01, 2017-09-21:
Be careful also with the use of lstrip("cid:"). It'll cause you trouble if you have the bad luck that the filename begins with "c,i or d" (see attached capture)
[image]

AKKenn, 2017-09-22:
Im having the same issue as @mariuszkobak. Every once in a while ill get this:

[6e60e7d8] filesystem access error: cannot open file /tmp/AlexaPi-runtime-1DrjWK/6403e59c-42f8-49ab-bece-949fd261fbae.c8feb6cb-3dcb-4bd2-a683-355238d10b61.0-0.mp3 (No such file or directory)
[6e60e7d8] core access error: File reading failed
[6e60e7d8] core access error: VLC could not open the file "/tmp/AlexaPi-runtime-1DrjWK/6403e59c-42f8-49ab-bece-949fd261fbae.c8feb6cb-3dcb-4bd2-a683-355238d10b61.0-0.mp3" (No such file or directory).
[63b00d58] core input error: open of `file:///tmp/AlexaPi-runtime-1DrjWK/6403e59c-42f8-49ab-bece-949fd261fbae.c8feb6cb-3dcb-4bd2-a683-355238d10b61.0-0.mp3' failed
[63b00d58] core input error: Your input can't be opened
[63b00d58] core input error: VLC is unable to open the MRL 'file:///tmp/AlexaPi-runtime-1DrjWK/6403e59c-42f8-49ab-bece-949fd261fbae.c8feb6cb-3dcb-4bd2-a683-355238d10b61.0-0.mp3'. Check the log for details.

Any idea's?

commit fec944ed (2017-10-08): fixing filenames to prevent bad characters (#314)

* fixes #309
closed 2017-10-08 by renekliment