alexa-pi/AlexaPi #316: Error when doing amazon autentication

GNetz, 2017-09-28:
Did everything correctly, but when i log in into the amazon account from auth_web.py it gives me this error

Any possible causes/solutions? Thanks

[image]

maso27, 2017-09-28:
Look at my post [here](https://github.com/alexa-pi/AlexaPi/issues/277#issuecomment-315896392).

maso27, 2017-09-28:
Also, you may want to edit your first post to remove your user-specific information.  Client ID, secret, etc.

And yes, you need a refresh token to use AlexaPi.

renekliment, 2017-10-14:
@GNetz Try to have those values without spaces. Otherwise, if you don't provide the log, there is nothing we can do.