alexa-pi/AlexaPi #319: Amazon Not Authorizing

eric8711, 2017-10-07:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
``` Jessie
```

**Your hardware platform and model you are running on:**
```
rpi 3
```

**Python release (`python2 --version`):**
```
python 2.7
```

**Description of problem:**
```
I have triple checked all the creds are right in the startup script and config file, but one I go to
I get this amazon error as shown in screen shot, I am able to log in with no issue
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
![image](https://user-images.githubusercontent.com/22085674/31303350-e71af5ea-aac0-11e7-8bea-df6458576eed.png)

```

eloverphx, 2017-10-23:
I get the same message and tried changing the info on Amazon a multitude of ways. I have tried installing Alexa using another method which uses port 3000 which then allowed to authenticate and get the token. Where can we change the port in this script to use port 3000?

adumont, 2017-10-28:
I have the same issue. How can I help?

adumont, 2017-10-28:
I managed to get through this. When the script asks for the "Enter your Device Type ID []", use the "Product ID" (the name you gave), not the "Amazon ID".

I think the question should be refrased.