alexa-pi/AlexaPi #325: Alexa crashes every time, after say "YES" with threading error

lucawow, 2017-10-09:
**Your OS (including version) where you are running AlexaPi:**
OrangePi Armbian Jessie

**Your hardware platform and model you are running on:**
OrangePi PC

**Python release (`python2 --version`):**
Python 2.7.9
**Description of problem:**
Alexa Start without problem and it say "Hello", then I say ALEXA, and Alexa Say "YES" 
so I ask a question but Alexa crashes every time with every word I say.  

File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
self.run()
File "/usr/lib/python2.7/threading.py", line 763, in run
self.__target(*self.__args, **self.__kwargs)
File "/opt/AlexaPi/src/main.py", line 542, in trigger_process
alexa_speech_recognizer(audio_stream)
File "/opt/AlexaPi/src/main.py", line 347, in alexa_speech_recognizer
resp = requests.post(url, headers=headers, data=data)
File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 112, in post
return request('post', url, data=data, json=json, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/requests/api.py", line 58, in request
return session.request(method=method, url=url, **kwargs)
File "/usr/local/lib/python2.7/dist-packages/requests/sessions.py", line 518, in request
resp = self.send(prep, **send_kwargs)

If I don't say a word, service remain UP and it write
INFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 21.03  7.88 -7.71 -8.57 -1.12  2.73 -3.77  0.49  5.58 10.63  5.69 10.76 11.52 >

**Expected:**
that it doesn't crashes

**
I read another thread with my question, but i' can't figure how resolve it
I have installed pulseaudio so now i use "pulse" like input and "pulse" like output, i hear alexa and alexa listen to me because it say "YES" so, how could I fix it? THANKS

renekliment, 2017-10-11:
Post the whole log.

renekliment, 2017-11-04:
Closing due to inactivity.

closed 2017-11-04 by renekliment