alexa-pi/AlexaPi #327: Custom Trigger Wake Word doesn't work

tkamaal, 2017-10-12:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Raspberry Pi OS Stretch 
VERSION_ID : "9"
VERSION = "9" (Stretch) 
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi Model B
```

**Python release (`python2 --version`):**
```
Python 2.7.13
```

**Description of problem:**
```
Aim : To change wake word from "alexa" to "something of my choice".  
```

**Expected:**
```
Pi should start responding to custom wake word set by me in the config.yaml file. 
```

**Problem-relevant `config.yaml` entries:**
```
Yes. 
pocketsphinx:
enabled: true
voice_confirm: true

phrase: "tom"
threshold: 1e-10
```

**Steps to reproduce:**

1. sudo nano etc/opt/AlexaPi/config.yaml
2. Edit the phrase in the file
3. Reboot the Pi or use sudo systemctl start AlexaPi.service

**Traceback (if applicable):**
```

```

**Additional info:**
```
I believe this can be done by modifying the "phrase " option in config.yaml file. This was suggested in one of the issues. I modified the wake word in config.yaml file and restarted the Pi but it didn't work after that. 
```

tkamaal, 2017-10-12:
#222

renekliment, 2017-10-12:
https://github.com/alexa-pi/AlexaPi/wiki/Triggers#voice-triggers
Read up. Try tweaking the threshold.

tkamaal, 2017-10-12:
Sure, I will try that.

tkamaal, 2017-10-12:
@renekliment  Doesn't seem to work even after playing with higher threshold values. Tried short wake words like "Tom" and "John" .  Tried threshold values 1e-50 and 1e-75.

renekliment, 2017-10-12:
Are you an English native speaker? The thing is sensitive to accents.

tkamaal, 2017-10-12:
I am not native English speaker. I asked my colleague who is an American and still doesn't seem to work . Threshold is set to 1e-50. 

When I say "alexa" it works fine. How does accent matter when we change the wake word? Please explain so it would be helpful for me to understand better.

renekliment, 2017-10-14:
I'm pretty sure it's a threshold / audio issue. Upload your recording here - the one that AlexaPi makes - see wiki on where to find that. This works otherwise.

DeepBlueSea123, 2017-11-03:
I'm unable to find AlexaPi.conf in /etc/opt

ShadeAJ, 2017-11-07:
Hi, I recently made the AlexaPi and wanted to also know how to change it to the phrase predetermined by amazon, "computer". This is something to add to the issue (it is by Github user bionemesis):

So according to the kitt.ai documentation, you can do it if you're using kitt.ai. But the documentation isn't clear on how. So I did some digging and figured it out. After getting everything up and running and responding to Alexa...

Visit https://snowboy.kitt.ai/ and register/login. Go through the process to create your own hotword. Naturally, you're going to use "Computer", right? Download the created "Computer.pmdl" file and rename it to "alexa.umdl". Stop wake word detection on your device and replace the file "~/Desktop/alexa-avs-sample-app/samples/wakeWordAgent/kitt_ai/snowboy/resources/alexa.umdl" with the renamed file you just downloaded (I backed up the existing file, just in case). Then restart wake word detection and you're good to go.

It, it, responds to computer... :)

Back to the Issue:

So I think this might be the way but I would really like someone to completely specify the steps in the process.

E3V3A, 2017-12-17:
Try with 3 syllable words. Obviously things like "Tom" is just one and become problematic with other words.
I would try with words like: `Tomato` (toh-ma-toh), `capito` (kah-pi-toh) etc.