alexa-pi/AlexaPi #329: I don't know why my AlexaPi failed

Linsanity77, 2017-10-14:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

renekliment, 2017-10-14:
Read the docs.
Read the logs.
You need to set up your audio.

Linsanity77, 2017-10-14:
Thanks, but how do I do that?

renekliment, 2017-10-14:
There are links to the docs in the README, which is the Github homepage, there is Github wiki, which is something you should always try and even the systemctl command tells you where it is.

Linsanity77, 2017-10-14:
Ok thank you