alexa-pi/AlexaPi #330: AlexaPi's Device are "offline" in the Alexa App

Raffix7, 2017-10-14:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Raspbian STRETCH WITH DESKTOP September 2017

**Your hardware platform and model you are running on:**
Raspberry Pi 3 Model B

**Python release (`python2 --version`):**
Last Version (October 2017)

**Description of problem:**
In the AlexaAPP my raspberry pi 3 with Alexa Pi installed is OFFLINE
The AlexaPi can't sync with the Alexa App. I want to have the same to-do list and same skills etc. but seems like AlexaPi and Alexa App work independently although I used the same account on both devices. I cant add skills from the store because when i try to enable it from the app, the skill isn t being installed on the Pi

**Expected:**
I want to synchronize the app and the AlexaPi to add new skills e synchronize the data

**Problem-relevant `config.yaml` entries:**
---

**Steps to reproduce:**

1. Add a skill from the app
2. Try to use that Skill on the AlexaPi
3. It doesn t exist

**Traceback (if applicable):**
--
**Additional info:**

--

renekliment, 2017-10-14:
1. https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#why-do-i-see-the-alexapi-device-offline-in-the-alexa-app--why-do-i-get-timers-and-alarms-are-not-currently-supported-on-this-device
2. People have been talking about skills in several issues. There doesn't seem to be a clear reason why it works for some people and for some, it doesn't

closed 2017-10-14 by renekliment

Raffix7, 2017-10-14:
so i can do somethings or not ? I can try to reinstall all the AlexaPi but should it work after this? It's very important for me because i want to use this as base of my exam's project.
before i dont tell that my amazon develop account is italian.. is this a problem? i set the location for the AlexaPi to US and i downloaded the app form american app store on IOS

renekliment, 2017-10-14:
You probably can't.
If you're doing everything correctly, you have all the things right in the _Alexa app_ and it doesn't work for you, then there's probably nothing you can do.

Location probably isn't a problem.

Raffix7, 2017-10-14:
how can i test in other way? i can create new amazon develop account maybe mine is bugged.. or recreate from 0 the alexapi device i dont know.. i just dont understand why to some people work correctly