alexa-pi/AlexaPi #331: AlexaPi doesn't show up in the Amazon Web App

Cole128, 2017-10-16:
**My OS (including version) where I am running AlexaPi:**
```
Raspian - from Debian Jessie
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 3 Model B
```

**Python release (`python2 --version`):**
```
2.7
```

**Description of problem:**
```
Upon going to the amazon alexa [web app](https://alexa.amazon.com/) and signing in, Amazon just wants me to setup a new Alexa device. It doesn't just show my dash for my RPi Alexa as it did before.
```

**Expected:**
```
That it will show the dashboard for my AlexaPi.
```

**Steps to reproduce:**

1.  Go to the amazon alexa web app.
2. See if you can access your AlexaPi device.

I understand this might be related to the fact that you guys are using the old API, but I still thought it would be good to see if anyone else is having this problem. Some more encouragement to use the new API at least! Thanks so much.

michibuddha, 2017-10-31:
Same here just finished building mine

Cole128, 2017-10-31:
Nice to hear it's not just me. I'm surprised the devs haven't answered this yet. :confused:

renekliment, 2017-11-01:
I've had my AlexaPi devices disappeared after using echoism.io which uses the new API. I have suspected that they might have just discontinued the old devices, but I've logged in now and I have them back.

Maybe there was a bug and it works again? Try now.

Also, if it doesn't, there is nothing we can do. It's totally up to Amazon.

Cole128, 2017-11-01:
Yeah, mine works again. Idk what happened, but I'm just glad it works. Would it be helpful to report this if it happens again later, or should just wait and only report if it lasts longer than x days?

Cole128, 2017-11-01:
Also maybe this could be motivation for a big upheaval to switch to the new API, if this is a user-wide issue.

Cole128, 2017-11-01:
I deregistered my echosim.io device from the web app, and now it just displays the setup page, as if I have no Alexa device at all.

Cole128, 2017-11-02:
Maybe I could setup up AlexaPi again, and then it would work? It depends if the web app is working for other users.

michibuddha, 2017-11-03:
Mine still isnt showing in app.

Cole128, 2017-11-03:
@michibuddha yes, but did you use echosim.io ? I just wonder if maybe that is what is causing the issue.

michibuddha, 2017-11-04:
Hey no bro. I build my AlexaPi the day I commented. I didn't knew what
echoism.io ia until you said something. For me maybe problem is Alexa voice
service just launched in India maybe that is the issue and its not snappy
as it supposed be. There is lag between question and answer like 5 to 10 sec

Cole128, 2017-11-04:
Ok it's good to know that echosim.io isn't the problem. That probably means that amazon has dropped support for the old API and AlexaPi NEEDS to switch to the new one. The slow response is an issue with AlexaPi, one that the devs are working on. I don't know if the fact that you are in India is affecting it.

michibuddha, 2017-11-05:
Yeah maybe. Btw we can try to switch to other Github alexa voice service
project for what it's worth. They're using the new APIs so we may get
lucky. There latest update was on 11th of October. I couldn't find when the
AlexaPi was updated last

Cole128, 2017-11-05:
What project is that?

michibuddha, 2017-11-05:
Amazon voice service aws sample app. Google it

Cole128, 2017-11-05:
Yeah I see it. It's pretty nasty to use though. To run it, you need to use three separate terminal windows each with a different command. AlexaPi is packaged more like an actual application, while the sample app is like a hacked-together demo. I'd rather stick with this, and get them to update to the new API.

michibuddha, 2017-11-05:
That's why I came here first rather than that project. Bit complex to
operate and will loose service after reboot each time manual start -_-

Cole128, 2017-11-05:
Yeah, it's pretty garbage. I'm just going to bug the devs here to switch to the new API.

michibuddha, 2017-11-05:
Better that way. It may happen soon I hope. I am in with you on this ride ☝

Cole128, 2017-11-05:
And I'm with you!

Cole128, 2017-11-05:
I think I will submit a feature request for it later. I'll keep you updated.

michibuddha, 2017-11-05:
Kudos

renekliment, 2017-11-05:
@Cole128

> I'm just going to bug the devs here to switch to the new API.

Don't. Just ... don't. If we had time to do that, it would have been done a long time ago.

Cole128, 2017-11-05:
Sorry. I didn't actually mean bug. You guys are working hard, and I love what you do. Would it be ok for me to reference this bug on #50 though?

renekliment, 2017-11-05:
@Cole128 I have my AlexaPi devices in the _Alexa app_. It's probably something you should report to Amazon.

Cole128, 2017-11-05:
@renekliment  Do they show up in the Alexa Web App though?

michibuddha, 2017-11-05:
I can see my device in Amazon alexa app. It's just offline. I can set
weather location and temperature units. But nothing else (maybe that's
limited to India cz its just launched here)
Btw I have posted the log you've asked for delay please see that.
Thx

Cole128, 2017-11-05:
If you can see your device in the app, then you aren't having this issue. And what log? Also, @renekliment I have no access to the alexa mobile app, so I have been using the web one.

Cole128, 2017-11-30:
I have reinstalled alexa and it still doesn't show up online on the web app.

gbCambridge, 2017-12-08:
Same here....my rpi shows in the list of Echo devices but is never online.  The old Amazon 3-terminal app was a pain, but it gave me control of my house automation devices. This app does not, but it is simpler to run and does not conflict with my DAC, as the old app did....with its wakeword implementation.  So, I seem to be stuck with this version but I have lost control of house devices...and it is always offline?  Something fishy here!

Cole128, 2017-12-08:
Yeah, @gbCambridge I think a lot of people have had the offline issue, search for it in the issues section of this github. My issue is different, my device doesn't show up at all. I just get the screen asking me to connect a device. I just worded my post above badly.

Cole128, 2017-12-17:
New development: My device still doesn't show up, but when I click to sign out of the web app, it appears in the background, with all its cards. But then I log out and can't use it, and if I log back in I am back at the welcome screen.

Here is what happens for a few seconds after I click sign out:
[image]

pfiol, 2018-01-10:
Same here @Cole128 

Unfortunately, I think that if you want AlexaPi to switch to the new API you will have to code it yourself... And even that doesn't mean you will see your device on the app as online....

Cole128, 2018-01-10:
Yeah. In the meantime I have an actual Alexa so I will play with that.

jabulonboaz, 2018-02-10:
Hello all any update on this? I have just build an AlexaPi and it works from vs the AVS sample app or the AssistansPi sample app because I want it on a Raspberry PI zero W as it’s light and running very well on Raspbian Jessie lite. It’s realy a shame not to be able to control my smart home devices with the AlexaPi ‘‘tis such a great project. 
I have the same issue like everyone else, AlexaPi appears in the Alexa app only as offline, i can configure the location ion based info like time and address but it won’t talk to any of my smarthome devices even if I disassociate my smart devices accounts and delete all of them and re associate AlexaPi won’t see them or talk to them.

dvbit, 2018-04-25:
So at least I could see , thanks to a slow connection on the mobile that every activity seems to be recorded on alexa.amazon.com (not co.uk).
However if I log in on web or mobile I get automatically redirectred to co.uk.
I discovered this is due to ip (I am in  from europe).
Using a VPN to exit with US IP indeed does not redirect me to the co.uk domain however there is no trace of the activities....
I am lost.

Could someone of the developers help? without the possibility of adding skills the whole thing is a (very nice) exercise but little more.
Thanks for the nice and sleek work btw.

mpearon, 2020-02-18:
Just adding my name to the list of people looking forward to new API support! :)

GeordieRoss, 2021-03-21:
Bump lmao