alexa-pi/AlexaPi #333: Microphone strikethrough when AlexaPi active

AirBurn1, 2017-10-25:
**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
```
cd ~ && wget https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh && bash alsa-info.sh
--2017-10-25 00:58:38--  https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.200.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.200.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27872 (27K) [text/plain]
Saving to: ‘alsa-info.sh.2’

alsa-info.sh.2                100%[================================================>]  27.22K  --.-KB/s   in 0.03s

2017-10-25 00:58:38 (977 KB/s) - ‘alsa-info.sh.2’ saved [27872/27872]

ALSA Information Script v 0.4.64
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  lsmod
  aplay
  amixer
  alsactl
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See 'alsa-info.sh --help' for command line options.

Automatically upload ALSA information to www.alsa-project.org? [y/N] :
```

**Your OS (including version) where you are running AlexaPi:**
```
Jessie v8
```

**Your hardware platform and model you are running on:**
```
Rpi 3 running magic mirror
```

**Python release (`python2 --version`):**
```
Python 2.7.9
```

**Description of problem:**
```
Microphone icon has a strikethrough and AlexaPi doesn't seem to be working
```

**Expected:**
```
Alexa to respond when I say "Alexa"
```

**Additional info:**
```
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Wed 2017-10-25 00:35:39 EDT; 28min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 614 (python)
   CGroup: /system.slice/AlexaPi.service
           └─614 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Oct 25 00:35:46 magicpi python[614]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Oct 25 00:35:46 magicpi python[614]: connect(2) call to /tmp/jack-999/default/jack_0 failed (err=No such file or directory)
Oct 25 00:35:46 magicpi python[614]: attempt to connect to server failed
Oct 25 00:35:53 magicpi python[614]: INFO: Checking Internet Connection ...
Oct 25 00:35:53 magicpi python[614]: INFO: Connection OK
Oct 25 00:35:53 magicpi python[614]: INFO: AVS token: Requesting a new one
Oct 25 00:35:53 magicpi python[614]: INFO: AVS token: Obtained successfully
Oct 25 00:35:55 magicpi python[614]: [6bf004d8] alsa audio output error: cannot estimate delay: Input/output error
Oct 25 00:36:34 magicpi python[614]: INFO: Triggered: pocketsphinx
Oct 25 00:37:26 magicpi python[614]: INFO: Triggered: pocketsphinx
```

stephenmelody, 2017-11-03:
I also have this issue, I have to keep stopping and starting the service to get it to work, or reboot the pi.

renekliment, 2017-11-04:
> Oct 25 00:36:34 magicpi python[614]: INFO: Triggered: pocketsphinx
> Oct 25 00:37:26 magicpi python[614]: INFO: Triggered: pocketsphinx

So it obviously triggers. Maybe it only doesn't play the _yes_ back.

> Microphone icon has a strikethrough

That's probably only AlexaPi eating your mic, because you don't have PulseAudio installed so multiple apps could use it at the same time. There is no issue there.

stephenmelody, 2017-11-04:
I get this a lot too... it just seems to bomb out after a while.

`Nov 04 08:04:50 Magic-Mirror python[454]: ERROR: URLError: [Errno 111] Connection refused
`