alexa-pi/AlexaPi #334: AlexaPi Service Error!

ConnorPlaysYT, 2017-10-28:
Anyone help me with this issue whenever i check the status of the AlexaPi service?
Error: https://hastebin.com/isofewepet.sql

leetwanker, 2017-10-29:
https://github.com/alexa-pi/AlexaPi/issues/309

renekliment, 2017-11-04:
Why did you ignore the issue template? I don't even know what OS version you have, nor the audio setup from the script.

ansu1997, 2020-03-09:
My HDMI is connected and I have connected headphones too ..
I checked #500 and did the correction like they told in media player.cpp but no change ...

ansu1997, 2020-03-09:
The audio just comes from HDMI and not through the jack no matter what I do and I’m running Alexa Sdk version on pi

deyabu, 2020-07-28:
Hi everybody

I have the same issue without audio from Alexapi when finished the installation.  Alexapi is listening very well with all logic status (Listening, Thinking, Speaking, Idle) but the audio not appeared by the three available outputs (HDMI, Analoge or USB Headphones) even changing these outputs by Raspberry audio menu not working. 

Suddenly, it occurred to me to ask Alexa to enable audio .... I Said "Alexa enable the audio" ...... and she did .... Eureka !!!! It this not a joke ... Now is working fine.   May be is a good news for the rest of people that have spend many time with this audio issue...KISS (Keep It Simple and Stupid)

Regards
Deyabu Design from Santiago, Chile