alexa-pi/AlexaPi #336: SSL error on AVS developed?

willjgi, 2017-10-31:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

MagicMarvMan, 2017-11-02:
Hi,

I think this does **not** work. Amazon automatically redirects you to the `https://` page.

You have to whitelist (i think) `https://localhost:{port}` in the return urls.

willjgi, 2017-11-03:
Hi MagicMarvMan - thanks for reply.
I have whitelisted the https:// versions - the problem is the program requests http:// as return url so amazon rejecting.
As I see it the code needs to set return url as https:// is that possible?