alexa-pi/AlexaPi #337: No reaktion

marcel7162, 2017-11-02:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
http://www.alsa-project.org/db/?f=70643e27441d034b8816415832fef271b4b81638

**Your OS (including version) where you are running AlexaPi:**

Jessi

**Your hardware platform and model you are running on:**
RP3

```

**Python release (``):**
Python 2.7.9

```

**Description of problem:**
Airplay no  Sound. Nö Reaktion if i say Alexa 

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

renekliment, 2017-11-04:
We should probably deprecate AirPlay support as it relies on people having set dmix or PulseAudio and config properly and I'm not sure it's something we can do reliably because of that.

Regarding no reaction ... ugh ... read the docs ...
first ... check the logs https://github.com/alexa-pi/AlexaPi/wiki/Debugging
second ... voice triggering is tricky ... play with it ... https://github.com/alexa-pi/AlexaPi/wiki/Triggers

maso27, 2017-11-06:
@renekliment 
I agree that Airplay can be removed, since it's really just a separate package that sits on top of the Alexa app.
If people want it they can pursue shairport-sync on their own.  It is pretty cool though if you use it to stream to multiple devices simultaneously.