alexa-pi/AlexaPi #339: Late feedback/answer from AWS

michibuddha, 2017-11-05:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```Raspbian - Stretch 
```

**Your hardware platform and model you are running on:**
```
Raspberry Pi 2
```

**Python release (`python2 --version`):**
```
2.7
```

**Description of problem:**
```i
I built my RPi echo last week its working but feedback is so late. Like 10 to 15 seconds for an answer. sometimes I think its not working at all. Its freezed after multiple questions & a reboot is required.
```

**Expected:**
```
It should work instantaneously or maybe a acceptable delay of 2, 3 seconds
```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

renekliment, 2017-11-05:
Post the log when running in debug mode.

Read up on the docs: https://github.com/alexa-pi/AlexaPi/wiki

michibuddha, 2017-11-05:
After stopping the Alexa service I got this log

2017-11-05 14:31:04 DEBUG: Setting up playback handler: VlcHandler
2017-11-05 14:31:04 INFO: Checking Internet Connection ...
2017-11-05 14:31:04 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-11-05 14:31:08 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1                                                                                                                                                             .1" 404 29
2017-11-05 14:31:08 INFO: Connection OK
2017-11-05 14:31:08 INFO: AVS token: Requesting a new one
2017-11-05 14:31:08 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2017-11-05 14:31:16 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/                                                                                                                                                             1.1" 200 945
2017-11-05 14:31:16 INFO: AVS token: Obtained successfully
Expression 'paInvalidSampleRate' failed in 'src/hostapi/alsa/pa_linux_alsa.c', l                                                                                                                                                             ine: 2048
Expression 'PaAlsaStreamComponent_InitialConfigure( &self->capture, inParams, se                                                                                                                                                             lf->primeBuffers, hwParamsCapture, &realSr )' failed in 'src/hostapi/alsa/pa_lin                                                                                                                                                             ux_alsa.c', line: 2719
Expression 'PaAlsaStream_Configure( stream, inputParameters, outputParameters, s                                                                                                                                                             ampleRate, framesPerBuffer, &inputLatency, &outputLatency, &hostBufferSizeMode )                                                                                                                                                             ' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 2843
2017-11-05 14:31:16 DEBUG: Stopping audio play
2017-11-05 14:31:16 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2017-11-05 14:31:16 DEBUG: Player State: State.Opening
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 66, in t                                                                                                                                                             hread
    self._capture.handle_init(self.AUDIO_RATE, self.AUDIO_CHUNK_SIZE)
  File "/opt/AlexaPi/src/alexapi/capture.py", line 113, in handle_init
    frames_per_buffer=chunk_size
  File "/usr/local/lib/python2.7/dist-packages/pyaudio.py", line 750, in open
    stream = Stream(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pyaudio.py", line 441, in __init_                                                                                                                                                             _
    self._stream = pa.open(**arguments)
IOError: [Errno -9997] Invalid sample rate

2017-11-05 14:31:16 DEBUG: Player State: State.Playing
2017-11-05 14:31:16 DEBUG: Started play.
2017-11-05 14:31:19 DEBUG: Player State: State.Ended
2017-11-05 14:31:19 DEBUG: Finished play.
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bl                                                                                                                                                             uealsa.device'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned err                                                                                                                                                             or: No such file or directory
ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or di                                                                                                                                                             rectory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bl                                                                                                                                                             uealsa.device'
[... cut ...]
-ceplen                 13              13
-cmn                    live            batch
-cmninit                40,3,-1         41.00,-5.29,-0.12,5.09,2.48,-4.07,-1.37,                                                                                                                                                             -1.78,-5.08,-2.05,-6.45,-1.42,1.17
-compallsen             no              no
-debug                                  0
-dict                                   /usr/local/lib/python2.7/dist-packages/p

michibuddha, 2017-11-10:
@renekliment I've posted log 5 days back. i know you are busy but can you take a look at it. Please? 
I want to use my AlexaPi daily and like every other human out there for each reply I dont have patience of 20 to 30 seconds.