alexa-pi/AlexaPi #342: AlexaPi will not verify

afastmovingstream, 2017-11-06:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
Raspbian
```

**Your hardware platform and model you are running on:**
```
RaspberryPi 3 
```

**Description of problem:**
```
I have tried many times to enter the http://10.xx.xx.205:5050 into a different web browser, but all the website does is say the website took too long to respond. I have tried a fresh install and still the same result and I have no idea what's going on.
```

willjgi, 2017-11-07:
I think this is an authorisation issue as avs wont allow http://  as return url which is what the token requester sets. I have similar problem and believe this is reason, unless someone can advise otherwise?

JuSSo81, 2018-09-11:
Hey, any anwser on this one?? Im in the same boat!