alexa-pi/AlexaPi #343: I don't know what's wrong

Linsanity77, 2017-11-07:
Hi, my AlexaPi still hasn't talked yet and when I look at its status, this comes up:

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sat 2017-10-14 17:04:06 EDT; 3 weeks 2 days ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 522 (python)
   CGroup: /system.slice/AlexaPi.service
           └─522 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Oct 14 17:04:06 raspberrypi systemd[1]: Starting Alexa client for all your d....
Oct 14 17:04:06 raspberrypi systemd[1]: Started Alexa client for all your de....
Oct 14 17:04:14 raspberrypi python[522]: INFO: Checking Internet Connection ...
Oct 14 17:04:14 raspberrypi python[522]: INFO: Connection OK
Oct 14 17:04:14 raspberrypi python[522]: INFO: AVS token: Requesting a new one
Oct 14 17:04:14 raspberrypi python[522]: INFO: AVS token: Obtained successfully
Oct 14 17:04:16 raspberrypi python[522]: [73e004d8] alsa audio output error:...r
Hint: Some lines were ellipsized, use -l to show in full.
can you please help me? Thanks.

AmitKKhanchandani, 2017-11-08:
Looks like you didn't setup the audio correctly, here are the steps https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging

can you post the complete log by executing main.py -l

Linsanity77, 2017-11-10:
Sorry, but for main.py -l it says the command is not found when I put it in terminal. Also, I don't know how to set it up using the wiki. Sorry I'm new to this.

AmitKKhanchandani, 2017-11-10:
No Problem!

Execute this  **/opt/AlexaPi/src/main.py -l** & Copy Paste the Log here

Linsanity77, 2017-11-10:
All 600 something lines?

Linsanity77, 2017-11-10:
And how do I copy all of it?

Linsanity77, 2017-11-10:
2017-11-07 13:37:04 INFO: Checking Internet Connection ...
2017-11-07 13:37:04 INFO: Connection OK
2017-11-07 13:37:04 INFO: AVS token: Requesting a new one
2017-11-07 13:37:05 INFO: AVS token: Obtained successfully
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 55, in thread
    inp = alsaaudio.PCM(alsaaudio.PCM_CAPTURE, alsaaudio.PCM_NORMAL, self._config['sound']['input_device'])
ALSAAudioError: Device or resource busy [plughw:CARD=Device,DEV=0]

[73b01728] alsa audio output error: cannot estimate delay: Input/output error

Here, I found it

renekliment, 2017-11-11:
@Linsanity77 You are still on an older version of AlexaPi. Please upgrade to 1.6 https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-do-i-upgrade-alexapi

Linsanity77, 2017-11-12:
Thank you, I have now updated it. I checked the status again and this comes up:

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sat 2017-11-11 13:39:44 EST; 3min 23s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 564 (python)
   CGroup: /system.slice/AlexaPi.service
           └─564 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Nov 11 13:39:48 raspberrypi python[564]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Nov 11 13:39:48 raspberrypi python[564]: ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
Nov 11 13:39:48 raspberrypi python[564]: Cannot connect to server socket err = No such file or directory
Nov 11 13:39:48 raspberrypi python[564]: Cannot connect to server request channel
Nov 11 13:39:48 raspberrypi python[564]: jack server is not running or cannot be started
Nov 11 13:39:52 raspberrypi python[564]: INFO: Checking Internet Connection ...
Nov 11 13:39:53 raspberrypi python[564]: INFO: Connection OK
Nov 11 13:39:53 raspberrypi python[564]: INFO: AVS token: Requesting a new one
Nov 11 13:39:53 raspberrypi python[564]: INFO: AVS token: Obtained successfully
Nov 11 13:39:55 raspberrypi python[564]: [718004d8] alsa audio output error: cannot estimate delay: Input/output error

What do I do now?

Alqio, 2017-11-15:
Hi, I get the same error. 
I tried debugging the audio with this guide: https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging 
However, on the "Checking if your system audio works" part when I'm testing recording, it requires your_input_device. How do I find your_input_device? Is it an ID? the device name? 

OP, have you tried that audio setup & debugging guide?

EmerickH, 2018-07-20:
Due to a large number of issues, this one is going to be closed because we think your problem is solved.
If you still have any problem, please read [this](https://emerickh.github.io/AlexaPi-config/setup/) first and feel free to re-open it!

closed 2018-07-20 by EmerickH

joshyeram, 2020-12-03:
Dec 03 04:40:22 raspberrypi python3[998]: CRITICAL: Your input_device 'plughw:1,0' is invalid. Use one of the following:
Dec 03 04:40:22 raspberrypi python3[998]: USB Microphone: Audio (plughw:1,0)
Dec 03 04:40:22 raspberrypi python3[998]: USB PnP Sound Device: Audio (plughw:2,0)

do I just put the input device as 'USB Microphone: Audio (plughw:1,0)'