alexa-pi/AlexaPi #345: Alexa works for around 10 minutes bit when i say 'Alexa' it wont respond

samiultahir, 2017-11-14:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
```

**Your hardware platform and model you are running on:**
```

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

jclarke0000, 2017-11-16:
I'm having a similar issue, but in my case Alexa freezes after my first utterance of "Alexa." I am able to ask my first question, but i get no response.  After this, AlexaPi seems to be frozen.  I need to reboot to get her back.

I think something has happened on Amazon's side of things.  Seems coincidental that Echo has just become available is some other countries -- Canada for one.  I think Amazon changed something that broke AlexaPi.

samiultahir, 2017-11-16:
Yeh probably.
I reall hope the issue can get fixed!

renekliment, 2017-11-16:
I don't see any log here. Read up on the docs: https://github.com/alexa-pi/AlexaPi/wiki

jclarke0000, 2017-11-18:
More on this.  I've tried upgrading to version 1.6, and I'm still getting the same experience.  Alexa responds when I first say Alexa, then does not answer my question.  Here is what I see in the log:

```
Nov 18 08:49:19 magicmirror systemd[1]: Starting Alexa client for all your devices...
Nov 18 08:49:19 magicmirror systemd[1]: Started Alexa client for all your devices.
Nov 18 08:49:24 magicmirror python[1521]: INFO: Magic Mirror HTTP Server - :5055
Nov 18 08:49:24 magicmirror python[1521]: INFO: Checking Internet Connection ...
Nov 18 08:49:24 magicmirror python[1521]: INFO: Connection OK
Nov 18 08:49:24 magicmirror python[1521]: INFO: AVS token: Requesting a new one
Nov 18 08:49:24 magicmirror python[1521]: INFO: AVS token: Obtained successfully
Nov 18 08:49:50 magicmirror python[1521]: Exception in thread Thread-6:
Nov 18 08:49:50 magicmirror python[1521]: Traceback (most recent call last):
Nov 18 08:49:50 magicmirror python[1521]: File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
Nov 18 08:49:50 magicmirror python[1521]: self.run()
Nov 18 08:49:50 magicmirror python[1521]: File "/usr/lib/python2.7/threading.py", line 763, in run
Nov 18 08:49:50 magicmirror python[1521]: self.__target(*self.__args, **self.__kwargs)
Nov 18 08:49:50 magicmirror python[1521]: File "/opt/AlexaPi/src/main.py", line 508, in trigger_process
Nov 18 08:49:50 magicmirror python[1521]: alexa_speech_recognizer()
Nov 18 08:49:50 magicmirror python[1521]: File "/opt/AlexaPi/src/main.py", line 318, in alexa_speech_recognizer
Nov 18 08:49:50 magicmirror python[1521]: process_response(resp)
Nov 18 08:49:50 magicmirror python[1521]: File "/opt/AlexaPi/src/main.py", line 409, in process_response
Nov 18 08:49:50 magicmirror python[1521]: with open(filename, 'wb') as f:
Nov 18 08:49:50 magicmirror python[1521]: IOError: [Errno 2] No such file or directory: '/tmp/AlexaPi-runtime-8i8nGR/65af0ab8-eedc-4f80-9e10-ff0e3c8a35a5#Alexa3P:1.0/2017/11/18/13/fe5f4d96aca646e7a8d0399b422c821b/49:47::TNIH_2V.2f9c9a51-f0f7-4565-b839-676c07c1114bZXV/0_36093145.mp3'
```

luismanson, 2018-01-18:
did you figure it out? im facing the same issue and vave no clue about it.

luismanson, 2018-01-18:
a little more of debug output

`2018-01-18 02:18:51 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com
2018-01-18 02:18:52 DEBUG: Start sending speech to Alexa Voice Service
2018-01-18 02:19:00 DEBUG: End recording
2018-01-18 02:19:00 DEBUG: Finished sending speech to Alexa Voice Service
2018-01-18 02:19:02 DEBUG: Processing Request Response...
2018-01-18 02:19:02 DEBUG: JSON String Returned: {
  "messageBody": {                                                                                                                                                          
    "directives": [                                                                                                                                                         
      {                                                                                                                                                                     
        "namespace": "Speaker",                                                                                                                                             
        "name": "SetMute",                                                                                                                                                  
        "payload": {                                                                                                                                                        
          "mute": false                                                                                                                                                     
        }                                                                                                                                                                   
      },                                                                                                                                                                    
      {                                                                                                                                                                     
        "namespace": "SpeechSynthesizer",                                                                                                                                   
        "name": "speak",                                                                                                                                                    
        "payload": {                                                                                                                                                        
          "contentIdentifier": "amzn1.as-ct.v1.Domain:Application:Knowledge#ACRI#85205af6-978b-40ad-a56c-eab028f0cb8c#Alexa3P:1.0/2018/01/18/05/e66b7bb2e56c4d5194b23762c0ce6da9/18:53::TNIH_2V.1ce99ac8-235e-476a-8ffe-734a15257072ZXV/0",                                                                                                             
          "audioContent": "cid:85205af6-978b-40ad-a56c-eab028f0cb8c#Alexa3P:1.0/2018/01/18/05/e66b7bb2e56c4d5194b23762c0ce6da9/18:53::TNIH_2V.1ce99ac8-235e-476a-8ffe-734a15257072ZXV/0_377173874"                                                                                                                                                      
        }                                                                                                                                                                   
      }                                                                                                                                                                     
    ]                                                                                                                                                                       
  },                                                                                                                                                                        
  "messageHeader": {}                                                                                                                                                       
}                                                                                                                                                                           
Exception in thread Thread-3:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/opt/AlexaPi/src/main.py", line 611, in trigger_process
    alexa_speech_recognizer(audio_stream)
  File "/opt/AlexaPi/src/main.py", line 420, in alexa_speech_recognizer
    process_response(resp)
  File "/opt/AlexaPi/src/main.py", line 510, in process_response
    with open(filename, 'wb') as f:
IOError: [Errno 2] No such file or directory: '/tmp/AlexaPi-runtime-ahq5I5/85205af6-978b-40ad-a56c-eab028f0cb8c#Alexa3P:1.0/2018/01/18/05/e66b7bb2e56c4d5194b23762c0ce6da9/18:53::TNIH_2V.1ce99ac8-235e-476a-8ffe-734a15257072ZXV/0_377173874.mp3'

^C2018-01-18 02:19:04 DEBUG: Cleaning up playback handler
`