alexa-pi/AlexaPi #348: Something strange came up in AlexaPi's status

Linsanity77, 2017-11-25:
Hello, I recently checked the status of my AlexaPi which I still have yet to successfully complete and this came up:

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sat 2017-11-11 13:45:45 EST; 7min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 551 (python)
   CGroup: /system.slice/AlexaPi.service
           └─551 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Nov 11 13:45:48 raspberrypi python[551]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Nov 11 13:45:48 raspberrypi python[551]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Nov 11 13:45:48 raspberrypi python[551]: ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
Nov 11 13:45:48 raspberrypi python[551]: Cannot connect to server socket err = No such file or directory
Nov 11 13:45:48 raspberrypi python[551]: Cannot connect to server request channel
Nov 11 13:45:48 raspberrypi python[551]: jack server is not running or cannot be started
Nov 11 13:45:53 raspberrypi python[551]: INFO: Checking Internet Connection ...
Nov 11 13:45:53 raspberrypi python[551]: INFO: Connection OK
Nov 11 13:45:53 raspberrypi python[551]: INFO: AVS token: Requesting a new one
Nov 11 13:45:53 raspberrypi python[551]: INFO: AVS token: Obtained successfully

I have never seen this before so I have no idea what this means or how to fix it. Please help me. Thank you

Linsanity77, 2017-11-25:
Now, something else came up instead and it's even more strange:

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sat 2017-11-11 13:45:45 EST; 43min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 551 (python)
   CGroup: /system.slice/AlexaPi.service
           └─551 /usr/bin/python /opt/AlexaPi/src/main.py --daemon

Nov 11 14:04:07 raspberrypi python[551]: self.run()
Nov 11 14:04:07 raspberrypi python[551]: File "/usr/lib/python2.7/threading.py", line 763, in run
Nov 11 14:04:07 raspberrypi python[551]: self.__target(*self.__args, **self.__kwargs)
Nov 11 14:04:07 raspberrypi python[551]: File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 77, in thread
Nov 11 14:04:07 raspberrypi python[551]: data = self._capture.handle_read()
Nov 11 14:04:07 raspberrypi python[551]: File "/opt/AlexaPi/src/alexapi/capture.py", line 119, in handle_read
Nov 11 14:04:07 raspberrypi python[551]: return self._handle.read(self._handle_chunk_size, exception_on_overflow=self._pa_exception_on_overflow)
Nov 11 14:04:07 raspberrypi python[551]: File "/usr/local/lib/python2.7/dist-packages/pyaudio.py", line 608, in read
Nov 11 14:04:07 raspberrypi python[551]: return pa.read_stream(self._stream, num_frames, exception_on_overflow)
Nov 11 14:04:07 raspberrypi python[551]: IOError: [Errno -9999] Unanticipated host error

renekliment, 2017-11-27:
> ```
> Nov 11 13:45:48 raspberrypi python[551]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
> Nov 11 13:45:48 raspberrypi python[551]: ALSA lib [...]

It's fine, just ignore it. We've switched the audio library and it outputs some stuff, nothing harmful.

closed 2017-11-27 by renekliment