alexa-pi/AlexaPi #349: i2S Mems microphone ics-43434 setup?

mano1979, 2017-11-25:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
raspbian
```

**Your hardware platform and model you are running on:**
raspberry pi

```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
i installed the microphone and it works when i use "arecord -Dmic_sv -c2 -r48000 -fS32_LE -twav -d10 -Vstereo test.wav" 
but how do i use it with alexapi? how to setup the config.yaml?

```

**Expected:**
Clear explaination on how to set audio in/out with examples

```

**Problem-relevant `config.yaml` entries:**
sound:
  # Name of your microphone device: "raspberrypi"
  # leave empty for default
  input_device: "default"

  playback_handler: "vlc"

  # you can go here with alsa, pulse or others
  # keep blank for the best audio output selected by the underlying code
  output: ""
  output device: "hw:0"
  # currently supports ALSA only
  output_device: "alsa"

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**

Status after alexapi start:

pi@raspberrypi:~ $ sudo systemctl restart AlexaPi.service
pi@raspberrypi:~ $ sudo systemctl status AlexaPi.service
● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: active (running) since Sat 2017-11-25 19:35:31 UTC; 33s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 1346 (python)
   CGroup: /system.slice/AlexaPi.service
           ├─1346 /usr/bin/python /opt/AlexaPi/src/main.py --daemon
           └─1361 /usr/bin/pulseaudio --start --log-target=syslog

Nov 25 19:35:34 raspberrypi python[1346]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Nov 25 19:35:34 raspberrypi python[1346]: ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
Nov 25 19:35:34 raspberrypi python[1346]: Cannot connect to server socket err = No such file or directory
Nov 25 19:35:34 raspberrypi python[1346]: Cannot connect to server request channel
Nov 25 19:35:34 raspberrypi python[1346]: jack server is not running or cannot be started
Nov 25 19:35:35 raspberrypi python[1346]: INFO: Checking Internet Connection ...
Nov 25 19:35:36 raspberrypi python[1346]: INFO: Connection OK
Nov 25 19:35:36 raspberrypi python[1346]: INFO: AVS token: Requesting a new one
Nov 25 19:35:37 raspberrypi python[1346]: INFO: AVS token: Obtained successfully
Nov 25 19:35:39 raspberrypi python[1346]: [6cb004d8] alsa audio output error: cannot estimate delay: Input/output error

```

revere521, 2017-11-30:
i have been using AlexaPi on my raspberrypi with pulseaudio in systemwide mode with no issues with my mic for quite a while (i use a ps-eye, which requires pulseaudio becase it has 4 mic channels) 

https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended

I don't think Alsa supports a stereo or multi-channel mic at all.

revere521, 2017-11-30:
may want to look at this too; https://www.raspberrypi.org/forums/viewtopic.php?t=173640

mano1979, 2017-12-01:
That is the tutorial i followed to install the mems mic. 
It works fine after following this tut. , when i call it manually.
Though somehow i can't seem to use it with alexapi.

This is a single channel microphone. (If you connect 2 of them you can set them up as a stereo mic)

revere521, 2017-12-02:
i can see in the log that pulseaudio and alsa appear to both be running; you may want to review the whole audio debugging guide in the wiki.  or read the pulse section and either set it up in systemwide mode or disable it.  https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#pulseaudio