alexa-pi/AlexaPi #351: AlexaPi and Device Recognition (Domoticz & HAB)

gbCambridge, 2017-12-06:
Hi,
I am running on rpi 3 on a 16GB card and Stretch version of OS. The rpi is hosting both Alexa and Squeezelite. The latter feeds through a DAC HAT board. So, Alexa voice comes from the rpi jack output and the Squeezelite music comes from the DAC. All OK , and works by setting the device to "dummy" in the config.yaml file...as it hits the pins used by the DAC when trying to flash lights (unnecessary).

So the good news is that Alexa is doing pretty well all that it should. In my house, there is also a setup of Domoticz which controls several devices. My two (official-real) Amazon dot devices can control the devices via the HAB which runs on another rpi. It might sound complex, but this all works...except that the Alexapi, when asked to turn on a device ALWAYS responds with "Sorry, I didn't find "name of device".  So it seems that it is attempting to trigger the device but the name recognition is always failing. The device that I am trying to control works perfectly through the Echo dot devices.
Can anyone advise ....am I taking AlexaPi too far right now?

Liamdelee, 2018-01-09:
There is a difference in the Echo devices and the AlexaPi (AVS setup). The AlexaPi cannot detect devices or controle devices because that is just not in the AVS but it is however in the echo devices. if this might change who knows, however you can solve this with awsiot described in a GitHub repository [here](https://github.com/alexa/alexa-avs-sample-app/issues/621) . just scroll down a bit and you will find a possible solution to make it work. Have fun :) (just found it myself)

gbCambridge, 2018-01-09:
It all works now. I simply had to use the latest Alexa from Amazon. There are several "guides" to setting up Alexa..but many use old versions. These do not communicate via HAB to Domoticz. 
In my Alexapi I managed to arrange that Alexa voice and high-quality music come from the same speakers, with the voice having its own volume control. That saved using a separate amplifier.  So, all is working...except that I have not yet implemented music selection from my HiFi via Alexa. There are a couple of attempts at this so I have signed up as a tester for the safest implementation.....it does not require opening a port on my modem.  Thanks for the help

Liamdelee, 2018-01-09:
what do you mean with latest alexa from Amazon? the AVS or??

gbCambridge, 2018-01-09:
Give me a while and I shall be precise. I have "discovered" a couple of rpis in a  cupboard so I am building another couple from scratch. One is a model 2 and the other is an original...this time I shall keep notes.
As far as I remember..I abandoned the many helpful articles and simply followed the current instructions from Amazon directly. It was cleaner and simpler.. the "helpful" articles are generally not updated :-(
George

Liamdelee, 2018-01-09:
okay,  yea i am thinking of using this repository as a start because it looks good maintained and stuff, but i also found the one from Amazon itself with a great guide, but no autostart. Which one would you recommend using? btw lucky you, i just bought 3 new once to tinker around with for some home automation (first time even touching a raspi :p)
thx again for the effort

gbCambridge, 2018-01-09:
I am feeling rather ashamed about this. I kept no notes...but I got very buried in the builds and various problems and solutions. In the various combinations that  I am running, they are headless and all have autostart. One is with a UPS HAT, so it runs 24/7 and monitors the battery and temperature. Another runs Logitech LMS as well as Alexa...they share the same amps and speakers. Another is a pure picoSqueezelite..no Alexa! and the version I am building now is a pure Alexa with just a mono speaker for use in my office.  The only bit of dev. work that is not working is the on/off button (wired on GPIO pins 5/6). It allows a soft shutdown and start...but only once!
There are many websites referring to rpi, Alexa, Logitech, Unix and so on. It becomes a management problem. May I suggest that you try out a product called Symbaloo?  It is free and allows you to manage pages of tiles, where each tile is a webpage link. I have several pages, for example rpi, Python, security, HiFi and so on... it is a great way to build a reference library.

gbCambridge, 2018-01-09:
Testing...but I think the build link is https://pimylifeup.com/raspberry-pi-alexa/

Will get back to you

gbCambridge, 2018-01-09:
Hi, I have only just finished the set up...on an rpi 2 this time. The link at https://pimylifeup.com/raspberry-pi-alexa/ is the correct one. It is lengthy but reliable. I am based on the Stretch version of the OS. This combination seems to be quite OK. It is running in headless mode with keyword wakeup.
The biggest problem I have is getting a good quality sound out of the jack output...but that should be minor.

Liamdelee, 2018-01-10:
hi there again, thanks for the info. I am looking into this one as well and it uses the code from the Alexa GitHub if i looked it up correctly. will be following that guide, but will also try and install the open source alexa-pi project and see if there are any differences and which one in my eyes is the best to go to (probably something for the weekend) will keep you updated as well.