alexa-pi/AlexaPi #353: Effects of an Echo Dot in the network on an rpi

gbCambridge, 2017-12-08:
In the notes, it states "AlexaPi will allow you to control your devices such as Hue provided you also have a genuine Echo in the network which acts as a gateway between the Alexa service and your Hue bridge."

In my network, I have two real Echo dots. Both of these can switch on devices via "Alexa, turn on HiFi" type of command.  The note above seems to imply that this should also be possible via the rpi installation....but it does not mention how this arises. Is it automatic or is there some binding that has to be done?

Apart from this lack..the AlexaPi installation is working well..it has run for two days without a problem, and I can also play back music via Squeezelite. So far I have not linked the two together because of the problem mentioned above.

revere521, 2017-12-10:
I have tplink wifi outlets; once my echo had identified them my AlexaPi could also control them - no additional setup was needed.  Not sure for folks with no amazon official devices.

gbCambridge, 2017-12-10:
Hi,
I have reverted back to the former 3-terminal Alexa. With that version, I can control everything. With the version here it never identified a device, even though I do have several official Amazon dots around the house. Not only that, I can run Squeexeelite simultaneously with Alexa through an on-board DAC. I think part of the problem is that this (simplified setup) version of Alexa is using old interfaces .. I suspect it is going to be useful staying with the original Amazon setup.