alexa-pi/AlexaPi #358: Error when logging in at localhost:5050

Cole128, 2017-12-17:
When I run the install script, or run auth_web.py individually, I get the amazon screen, but after clicking okay, I get an error:
[image]

HansMees, 2018-01-02:
I have the same problem

jroza1, 2018-01-03:
So then, what is the way forward?  Do we have to go back to the legit Amazon code without the autologin?  
I am getting the same error, but when I go into my console, I do see that my AlexaPi is registered.  Also as I call out to Alexa, she doesn't seem to be doing anything.

Cole128, 2018-01-05:
I think the only thing to do is wait for the devs to fix it. 🙁

Bibelo, 2018-02-10:
So I basically lost 3 hours because the dev didn't bother mentionning it on his page.

Thank you dev.

renekliment, 2018-02-10:
The old API is still supported.

Bibelo, 2018-02-10:
Thank you for your answer!

I have no way of knowing what's wrong. I've run the script at least 15 times, with variations. I tried creating a new object. That was looong because the thing tries to compile and install *every single time*

Bibelo, 2018-02-19:
Still no answer from the dev, while some users have managed to make it work without giving the solution.

"The old API is still supported" does not help me.

s1h4d0w, 2018-02-26:
This does seem to be a problem on Amazon's end. I have the exact same issue, I just commented on an older issue: https://github.com/alexa-pi/AlexaPi/issues/208#issuecomment-368525636

Everything is set up as it should be on my side, but when trying to allow the app on the Amazon website it just gives the error OP posted.

EmerickH, 2018-07-20:
Due to a large number of issues, this one is going to be closed because we think your problem is solved.
If you still have any problem, please feel free to re-open it!

closed 2018-07-20 by EmerickH