alexa-pi/AlexaPi #359: Audio hardware problem

mattlindop, 2017-12-20:
I'm trying to convert my Google AIY voice kit into AlexaPi but have run into problems with the audio hardware (I think). 

Audio debugging output here
http://www.alsa-project.org/db/?f=72335562cd34e06633d62e38b5ede39db8915d4f

Here's what I get from journalctl:
Dec 20 22:26:52 raspberrypi systemd[1]: Started Alexa client for all your devices.
Dec 20 22:27:01 raspberrypi pulseaudio[772]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Dec 20 22:27:01 raspberrypi pulseaudio[772]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon without a $DISPLAY for X11
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Dec 20 22:27:01 raspberrypi python[480]: ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Dec 20 22:27:01 raspberrypi python[480]: connect(2) call to /tmp/jack-999/default/jack_0 failed (err=No such file or directory)
Dec 20 22:27:01 raspberrypi python[480]: attempt to connect to server failed
Dec 20 22:27:04 raspberrypi python[480]: INFO: Checking Internet Connection ...
Dec 20 22:27:04 raspberrypi python[480]: INFO: Connection OK
Dec 20 22:27:04 raspberrypi python[480]: INFO: AVS token: Requesting a new one
Dec 20 22:27:05 raspberrypi python[480]: INFO: AVS token: Obtained successfully

Running latest Raspberian on a RP3

Any help would be gratefully received!

EmerickH, 2018-07-19:
A start of solving:
Your config is wrong: you have to setup your audio PCMs:
List them with `aplay -L` (for output) and `arecord -L` (for input) and change it in the config:
```
output: "alsa"
output_device: "plughw:YOUROUTPUTDEVICE"
input_device: "plughw:YOURINPUTDEVICE"
```

EmerickH, 2019-01-07:
@mattlindop You don't replied so I suppose that your problem is fixed.
If you still have the issue, please re-open this issue and provide more information please.

Emerick

closed 2019-01-07 by EmerickH