alexa-pi/AlexaPi #361: 400 Bad Request

TroutSherpa, 2017-12-25:
Being a nube to this I've read through the wiki in order to try and figure out what I've messed up.  I've checked my Allowed origins and Allowed return URLs and cannot seem to find my mistake.  When I tried to authenticate by going to my ipaddress:5050, I received a "400 Bad Request" error from Amazon.  Running a status check on AlexaPi i got the following return:

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled)
   Active: failed (Result: exit-code) since Sun 2017-12-24 20:18:02 EST; 1h 30min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 525 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 525 (code=exited, status=1/FAILURE)

Dec 24 20:17:34 raspberrypi python[525]: ALSA lib pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
Dec 24 20:17:34 raspberrypi python[525]: ALSA lib pcm_dmix.c:1022:(snd_pcm_dmix_open) unable to open slave
Dec 24 20:17:34 raspberrypi python[525]: Cannot connect to server socket err = No such file or directory
Dec 24 20:17:34 raspberrypi python[525]: Cannot connect to server request channel
Dec 24 20:17:34 raspberrypi python[525]: jack server is not running or cannot be started
Dec 24 20:18:01 raspberrypi python[525]: INFO: Checking Internet Connection ...
Dec 24 20:18:01 raspberrypi python[525]: INFO: Connection OK
Dec 24 20:18:01 raspberrypi python[525]: CRITICAL: AVS refresh_token not found in the configuration file. Run the setup a...ions).
Dec 24 20:18:02 raspberrypi systemd[1]: AlexaPi.service: main process exited, code=exited, status=1/FAILURE
Dec 24 20:18:02 raspberrypi systemd[1]: Unit AlexaPi.service entered failed state.
Hint: Some lines were ellipsized, use -l to show in full.

Any suggstions would be greatly appreciated.  I have greater hopes for this RPi than a retro Mario game.

revere521, 2018-01-01:
This looks like the setup didn't complete correctly or your developer tokens from amazon are missing - try running the setup again?

TroutSherpa, 2018-01-03:
I ran the setup again, and still received the previous response which is below:

Error Summary
400 Bad Request
The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.
Request Details
scope=alexa%3Aall
redirect_uri=http%3A%2F%2F192.168....
response_type=code
client_id=amzn1.application-oa2-client.20cc8bc9d85d4e179c....

It says in the Error Summary that the "redirect URI" is not "whitelisted".  What is it referring to as the URI?  Haven't come across that term before.  

Thanks for your time and help.  One day it'll make sense.

keithdadkins, 2018-01-10:
Same problem here. I am assuming that Amazon now forces https which this project does not implement?

ihorbond, 2018-03-10:
I received the same error. I logged into my amazon dev console, updated the return url again to the http://localhost:5050/code saved and closed tab. it worked afterwards and went to auth screen

ihorbond, 2018-03-10:
make sure you add all 4 URLs to allowed origins as per installation instructions

EmerickH, 2018-07-19:
```
pcm.c:2239:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
```
Your config is wrong: you have to setup your audio PCMs:
List them with `aplay -L` (for output) and `arecord -L` (for input) and change it in the config:
```
output: "alsa"
output_device: "plughw:YOUROUTPUTDEVICE"
input_device: "plughw:YOURINPUTDEVICE"
```

EmerickH, 2019-01-07:
See #366

closed 2019-01-07 by EmerickH