alexa-pi/AlexaPi #362: Pulse Audio Cookie file

ypeeps, 2017-12-29:
Thank you for reporting an issue with **AlexaPi**.

**Your OS (including version) where you are running AlexaPi:**
Raspbian GNU/Linux 9

**Your hardware platform and model you are running on:**
RaspberryPi 3

**Description of problem:**
I installed Pulse audio per the set up instructions and now Alexa won't start. I am getting an error in the log file. It seems that the pulse audio is looking for some sort of cookie file that wasn't mentioned in the wiki.

Logs begin at Thu 2016-11-03 12:16:43 CDT, end at Fri 2017-12-29 09:08:04 CST. --
Dec 29 09:03:02 raspberrypi systemd[1]: Started Alexa client for all your devices.
**Dec 29 09:03:11 raspberrypi pulseaudio[822]: [pulseaudio] authkey.c: Failed to open cookie file '/var/lib/AlexaPi/.config/pulse/cookie': No such file or directory
Dec 29 09:03:11 raspberrypi pulseaudio[822]: [pulseaudio] authkey.c: Failed to load authentication key '/var/lib/AlexaPi/.config/pulse/cookie': No such file or directory
Dec 29 09:03:11 raspberrypi pulseaudio[822]: [pulseaudio] authkey.c: Failed to open cookie file '/var/lib/AlexaPi/.pulse-cookie': No such file or directory
Dec 29 09:03:11 raspberrypi pulseaudio[822]: [pulseaudio] authkey.c: Failed to load authentication key '/var/lib/AlexaPi/.pulse-cookie': No such file or directory**
**Dec 29 09:03:11 raspberrypi pulseaudio[822]: [pulseaudio] server-lookup.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-da
Dec 29 09:03:11 raspberrypi pulseaudio[822]: [pulseaudio] main.c: Unable to contact D-Bus: org.freedesktop.DBus.Error.NotSupported: Unable to autolaunch a dbus-daemon with**
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.front.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround40.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround50
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround51
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround71.0:CARD=0'
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround71
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM iec958
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.iec958.0:CARD=0,AES0=4,AES1=130,AES2=0,AES3
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM spdif
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.hdmi
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.modem
Dec 29 09:03:11 raspberrypi python[448]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.phoneline
[... 14 lines cut ...]
Dec 29 09:03:12 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
[... cut ...]
## Configuration file for PulseAudio clients. See pulse-client.conf(5) for
## more information. Default values are commented out.  Use either ; or # for
## commenting.

; default-sink =
; default-source =
; default-server =
; default-dbus-server =

; autospawn = no
; daemon-binary = /usr/bin/pulseaudio
; extra-arguments = --log-target=syslog

; cookie-file =

; enable-shm = yes
; shm-size-bytes = 0 # setting this 0 will use the system-default, usually 64 MiB

; auto-connect-localhost = no
; auto-connect-display = no

renekliment, 2017-12-30:
You didn't install system-wide PA correctly.
`systemctl status pulseaudio`

ypeeps, 2017-12-30:
● pulseaudio.service
   Loaded: loaded (/etc/systemd/system/pulseaudio.service; disabled; vendor preset: enabled)
   Active: inactive (dead)

Dec 30 10:44:02 raspberrypi systemd[1]: [/etc/systemd/system/pulseaudio.service:1] Unknown section 'unit'. Ignoring.

revere521, 2018-01-01:
I would look at /etc/systemd/system/pulseaudio.service to see if there is a typo. Make sure the brackets are around the section heading - i.e. "[Unit]"

EmerickH, 2019-01-07:
@ypeeps You don't replied so I suppose that your problem is fixed.
If you still have the issue, please re-open this issue and provide more information please.

Emerick

closed 2019-01-07 by EmerickH