alexa-pi/AlexaPi #364: AVS refresh_token not found in config

QuantumDoge, 2018-01-07:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Raspbian Desktop 
4.9.59-v7+
**Your hardware platform and model you are running on:**
Raspberry Pi 3 Model B
**Python release (`python2 --version`):**
Python 2.7.13
**Description of problem:**
When checking is the service is running, it replies with the critical error "AVS refresh_token not found in the configuration file."
**Expected:
Should the token be there?
**Problem-relevant `config.yaml` entries:**
When checking the config file, the refresh token section is blank.
**Steps to reproduce:**

1. Run setup script
2. Enter correct Amazon developer information
3. Service does not start

**Traceback (if applicable):**
```

```

**Additional info:**

gunnicus, 2018-02-27:
I think I had same issue: https://github.com/alexa-pi/AlexaPi/issues/381

You need to leave a white space in front of values in config file, like below:
[image]

EmerickH, 2018-07-19:
Due to a large number of issues, this one is going to be closed because we think your problem is solved.
If you still have any problem, please feel free to re-open it!

closed 2018-07-19 by EmerickH