alexa-pi/AlexaPi #365: Audio input issues

xinyuehan, 2018-01-07:
**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**
http://www.alsa-project.org/db/?f=e61c32dd0bfa30048549df1a3a9a2e5e46ed4138

**Your OS (including version) where you are running AlexaPi:**
Raspbian 9

**Your hardware platform and model you are running on:**
Raspberry PI 3

**Python release (`python2 --version`):**
2.7.13

**Description of problem:**
I can hear alexa say hello when starting the service but she can't hear me. I tried to test my microphone using arecord but it's also silent. Tried running alexa in debug mode and checking the recordings but none seem to be created. Hoping someone could give me some advice on how to treoubleshoot this? I'm using a mini usb microphone plugged into the raspberry pi.

Liamdelee, 2018-01-10:
since you said it is also silent using arecord, i would suggest testing your microphone on a laptop or such just to make sure the mic is not broken. If it is not broken you will probably need to increase the input sensitivity (or something like that) in your raspberry pi. this can be found [here](https://pimylifeup.com/raspberry-pi-alexa/), and search for the Improving your Microphone Gain. hope this helps you forward a bit

TheHerodes, 2018-01-13:
I have the same problem and many others.I think that the mic is not listening at all.When i use common commands to record it give errors.(arecord t.wav)(arecord: main:788: audio open error: No such file or directory)

seannervt, 2018-01-28:
I either get an error saying the resource is busy or nothing at all when I try to test the mic. 

The debugging script 
http://www.alsa-project.org/db/?f=8386e3254f2355dbd1b3458b2a658624b6e03360

Your OS (including version) where you are running AlexaPi:
PRETTY_NAME="Raspbian GNU/Linux 8 (jessie)"
NAME="Raspbian GNU/Linux"
VERSION_ID="8"
VERSION="8 (jessie)"
ID=raspbian
ID_LIKE=debian

Your hardware platform and model you are running on:
Raspberry Pi 2

Python release (python2 --version):
Python 2.7.9 (default, Sep 17 2016, 20:26:04)

EmerickH, 2018-07-19:
Your config is maybe wrong: you have to setup your audio PCMs:
List them with `aplay -L` (for output) and `arecord -L` (for input) and change it in the config:
```
output: "alsa"
output_device: "plughw:YOUROUTPUTDEVICE"
input_device: "plughw:YOURINPUTDEVICE"
```