alexa-pi/AlexaPi #366: Fix for 400 Bad Request

patricklyons, 2018-01-21:
Fix for the "400 Bad Request" error message when trying to authenticate using http://localhost:5050.  This is happening because Amazon requires SSL which isn't supported with this version of AlexaPi.  The alternate installation method (https://github.com/alexa/alexa-avs-sample-app) does however support SSL - if you install this you can find the refresh_token that AlexaPi needs to run. 

1) Follow instructions to install alexa-avs-sample-app (https://github.com/alexa/alexa-avs-sample-app). 
2) After setup is complete open the refresh_tokens file (sudo nano /home/pi/Desktop/alexa-avs-sample-app/samples/companionService/refresh_tokens).  You'll see a line that starts with "refresh_token:"   The token begins at "Atzr...." (ignore the string of numbers prior to Atzr) and is quite long. To copy it I had to scroll to the right, hit return for new line, repeat repeat repeat until I could see it all on one screen.  Paste it into a text editor to get rid of all the line breaks and copy it to your clipboard. 
3) Follow the instructions to install AlexaPi.  At the end it will prompt you to go to http://localhost:5050, don't bother, it won't work, just close the installer with control c. 
4) Insert the refresh_token into the configuration file for AlexaPi (sudo nano /etc/opt/AlexaPi/config.yaml).  Should look something like this: 
refresh_token: "Atzr | amillionmoreletters" 
5) Start the service with "sudo systemctl start AlexaPi.service"

ihorbond, 2018-03-10:
I only followed the instructions described here
My fix for 400 error was:
1. Add http://10.0.1.26:5050/code to Allowed return URLs
2. Go to http://your_pi_ip:5050
3. Authorize app
4. Receive 500 bad request. Ignore
5. Go to http://your_pi_ip:5050
Since you already authorized your app the url will contain the refresh_token (parameter C). Copy and paste it into config.yaml

ihorbond, 2018-03-13:
I was able to get refresh_token doing this:
1. add http://localhost:5050 to allowed origins
2. add http://localhost:5050/code to allowed redirect urls
3. if your install script still running from your raspberry pi goto step 4
else re launch the server by running sudo ./auth_web.py from /opt/AlexaPi/src
4. navigate to http://localhost:5050 from your raspberry pi
5. I didn't get 400 error here. I saw authorization screen. Authorize. Receive 500 bad request
6. In the url there is authorization code that looks like this ANEctBuBjzro. Copy it.
7. Use your fav rest client(insomnia, postman etc) to send post request to https://api.amazon.com/auth/o2/token as described here [https://developer.amazon.com/docs/alexa-voice-service/authorize-companion-site.html](Authorization Code Grant). 
Set code param to code you copied at step 6 and redirect_url to http://localhost:5050/code
8. Send POST request. You should receive refresh token in the response. Paste it to config.yaml and restart alexa service.

acsrikar279, 2018-04-15:
@ihorbond 
Hello,
Can you please explain how to send request using Insomnia?
There are a lot of sections like body, header, query.. 
Thanks in advance

ihorbond, 2018-04-18:
@acsrikar279 send POST request with body type Form URL Encoded and Content-Type header set to application/x-www-form-urlencoded as you can see at https://developer.amazon.com/docs/alexa-voice-service/authorize-companion-site.html

EmerickH, 2018-07-20:
In some cases, it's working without any problem, we need to investigate more...

yugn27, 2018-07-20:
From 01 jan 2018 the AVS Java Sample App is in maintenance mode.

EmerickH, 2019-01-07:
Hello,
are you still experiencing issues with this?

patricklyons, 2019-01-07:
I actually just came up with this solution to see if I could make it
work....but have since moved on to other projects so I'm sorry, I can't
tell you if the workaround is still effective

shaukathussain, 2020-03-07:
I am facing the same problem on Raspberry PI 4B. here are the logs. Server can say "Hello" and also the speech is sent to the server from my Raspberry Pi4, but the HTTP status code 400 is sent back. anybody can help that will be highly appreciated. thanks

`2020-03-07 18:02:35 INFO: Triggered: pocketsphinx
2020-03-07 18:02:35 DEBUG: Stopping audio play
2020-03-07 18:02:35 DEBUG: Playing audio: /opt/AlexaPi/src/resources/alexayes.mp3
2020-03-07 18:02:35 DEBUG: Player State: State.Opening
2020-03-07 18:02:35 DEBUG: Player State: State.Playing
2020-03-07 18:02:35 DEBUG: Started play. 
2020-03-07 18:02:35 DEBUG: Player State: State.Ended
2020-03-07 18:02:35 DEBUG: Finished play. 
2020-03-07 18:02:35 DEBUG: Recording: Setting up
2020-03-07 18:02:35 DEBUG: Recording: Start
2020-03-07 18:02:35 DEBUG: Starting new HTTPS connection (1): access-alexa-na.amazon.com:443
2020-03-07 18:02:36 DEBUG: Start sending speech to Alexa Voice Service
2020-03-07 18:02:43 DEBUG: Recording: End
2020-03-07 18:02:43 DEBUG: Finished sending speech to Alexa Voice Service
2020-03-07 18:02:44 DEBUG: Processing Request Response...
2020-03-07 18:02:44 INFO: (process_response Error) Status Code: 400
INFO: cmn_live.c(120): Update from < 64.23 -4.51  4.28  6.26 -2.93  6.21 -1.25 -4.03  0.15 11.26  5.20  2.92  2.75 >
INFO: cmn_live.c(138): Update to   < 64.69 -4.74  4.36  5.85 -2.91  5.84 -0.56 -4.65  0.60 10.50  5.56  2.79  2.32 >
INFO: kws_search.c(656): kws 0.20 CPU 0.215 xRT
INFO: kws_search.c(658): kws 3.72 wall 4.092 xRT
`