alexa-pi/AlexaPi #371: Alexa say "Hello" but doesn't answer

lcouvreur, 2018-02-05:
Hello,

I'm trying to setup Alexa. When the service is launched now I can heard Alexa saying "Hello" but she didn't answer yet.

I had the following message when I run in debug mode `/opt/Alex*/src/main.py -d
` I get the following error : 
```
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.bcm2835.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
```
I checked in the issue and I modified the file `~/.asoundrc` following [the instruction here ](https://github.com/alexa/avs-device-sdk/issues/108#issuecomment-324738205)
But the problem remains after the modification of the file. 

I verified my microphone with `rec temp.wav` and my microphone is well recording. 
When Alexa is running in debug mode I can also see additional line when I make some noise. 

I have also the following error message when I check the status 
```
fév 05 22:30:00 raspberrypi python[1236]: ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'defaults.bluealsa.device'
fév 05 22:30:00 raspberrypi python[1236]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
fév 05 22:30:00 raspberrypi python[1236]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
fév 05 22:30:00 raspberrypi python[1236]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
fév 05 22:30:00 raspberrypi python[1236]: connect(2) call to /tmp/jack-999/default/jack_0 failed (err=No such file or directory)
```
Any idea to help me ?

Sorry I'm still a newbie, I only began to play with Raspberry/Arduino/ESP8266/Linux few weeks ago thanks to a course about IoT on Coursera. 

Thanks a lot

Lio

SoundGuy, 2018-02-08:
I'm having  a similar problem on Rpi b+
Alexa wakes up says hello but doesn't respond to me

NFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2018-02-08 19:19:23 DEBUG: Setting up playback handler: VlcHandler
2018-02-08 19:19:25 INFO: Checking Internet Connection ...
2018-02-08 19:19:25 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2018-02-08 19:19:30 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2018-02-08 19:19:30 INFO: Connection OK
2018-02-08 19:19:30 INFO: AVS token: Requesting a new one
2018-02-08 19:19:30 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2018-02-08 19:19:36 DEBUG: https://api.amazon.com:443 "POST /auth/o2/token HTTP/1.1" 200 945
2018-02-08 19:19:36 INFO: AVS token: Obtained successfully
2018-02-08 19:19:36 DEBUG: Stopping audio play
2018-02-08 19:19:36 DEBUG: Playing audio: /opt/AlexaPi/src/resources/hello.mp3
2018-02-08 19:19:36 DEBUG: Player State: State.Opening
2018-02-08 19:19:37 DEBUG: Player State: State.Playing
2018-02-08 19:19:37 DEBUG: Started play.
2018-02-08 19:19:40 DEBUG: Player State: State.Ended
2018-02-08 19:19:40 DEBUG: Finished play.
INFO: cmn_live.c(88): Update from < 41.00 -5.29 -0.12  5.09  2.48 -4.07 -1.37 -1.78 -5.08 -2.05 -6.45 -1.42  1.17 >
INFO: cmn_live.c(105): Update to   < 47.70 -4.21  2.05 -8.28 -3.60  2.93 -4.67 -7.54 -9.90 -3.31  1.11 -0.70  1.57 >
INFO: cmn_live.c(88): Update from < 47.70 -4.21  2.05 -8.28 -3.60  2.93 -4.67 -7.54 -9.90 -3.31  1.11 -0.70  1.57 >
INFO: cmn_live.c(105): Update to   < 45.78 -3.91  2.61 -7.68 -4.50  1.13 -3.77 -7.22 -9.18 -3.42  1.13  0.33  1.29 >

SoundGuy, 2018-02-08:
my alsa is 
 http://www.alsa-project.org/db/?f=6ba5dbda0ee4ac6d3cae98ea84f8660a18896414

lcouvreur, 2018-02-11:
SoundGuy did you find the solution ?

I'm still looking but without any success.
When I follow the process "[Running PA in system-wide mode (recommended)](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended)" but it don't solve my problem even worse : Alexa status is failed if I don't set the config.yaml like that
`input_device: "pulse"
output: "pulse"
output_device: ""`

I have to set input_device as "" to get an "active" status.

lcouvreur, 2018-02-11:
My Alsa debugging datas : http://www.alsa-project.org/db/?f=d6edb52c9e652de8cdd62cd812e974d3887273e3

SoundGuy, 2018-02-21:
@lcouvreur  i did not.

feh123, 2018-05-07:
Hi I have a similar issue. Alexa says hello but then does not answer. My debug file is as SoundGuy but ends at finished play. Any news on the solution? I wonder about my mic. I have made wav files using record but need to get them from my Pi to my PC - not sure how to do this!

EmerickH, 2018-07-19:
You have to set your `input_device` and `output_device` by finding the `plughw:SOMETHING` of your card in the output of `arecord -L` (for input) and `aplay -L` (for output).
You can also follow this:
https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended
(with SoX at the end if you are on a Raspberry)

EmerickH, 2018-07-19:
Due to a large number of issues, this one is going to be closed because we think your problem is solved.
If you still have any problem, please feel free to re-open it!

closed 2018-07-19 by EmerickH