alexa-pi/AlexaPi #372: Respeaker 2mic hat issue - no audio

jabulonboaz, 2018-02-06:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Raspbian Jessie light the latest build November 2017
**Your hardware platform and model you are running on:**
Raspberyyi Pi 3 ver B and Respeaker 2mic HAT  

**Python release (`python2 --version`):**
do not know i followed the instructions for this git to the letter 
**Description of problem:**
I have no audio - i just finished setup and there is no audio input or output, Th respeker HAT was tested before the build with it's drivers and the speaker and mic where working fine, 

**Expected:**
suggestions for resolution
**Problem-relevant `config.yaml` entries:**
could be but i am not sure how to define my Respeaker 2mic hat in the config.yalm audio part,

**Steps to reproduce:**
you need to have a respeaker 
**Traceback (if applicable):**
``
**Additional info:**
I am reading the  alexa pi git trying to figure out what is going on and running through the audio debug procedure but i got nowhere., I get no sound and i have implemented other assistants like "assistantsPi" with the same HW and same drivers for respeaker and same asound.conf file and worked but "AlexaPi" does not. Is the normal behaviour for the Alexa pi t o autostart on boot and say "Hello" right? but in my case i hear nothing.
Bellow is my asound.conf file and also i have blacklisted the integrated Pi sound-card so not only Respeaker speed2mice appears as card 0 device 0 so the bellow should be working. I also tried deleting the contents of "asoundrc" but no difference now i got "asound.conf" and "asoundrc" identical see file bllow. Maybe i need to edit /etc/opt/AlexaPi/config.yaml and add my input device? but i have no clue how to describe my  Respeaker as I do not know what part of the aplay -l description to use. do you know?
This is the "aplay -i" content:
card 0: seeed2micvoicec [seeed-2mic-voicecard], device 0: bcm2835-i2s-wm8960-hifi wm8960-hifi-0 []
  Subdevices: 1/1
  Subdevice #0: subdevice #0

BTW This is my asound.conf file:
# The IPC key of dmix or dsnoop plugin must be unique
# If 555555 or 666666 is used by other processes, use another one
pcm.!default {
    type asym
    playback.pcm "playback"
    capture.pcm "capture"
}

pcm.playback {
    type plug
    slave.pcm "dmixed"
}

pcm.capture {
    type plug
    slave.pcm "array"
}

pcm.dmixed {
    type dmix
    slave.pcm "hw:0,0"
    ipc_key 555555
}

pcm.array {
    type dsnoop
    slave {
        pcm "hw:0,0"
        channels 2
    }
    ipc_key 666666
}﻿`

jabulonboaz, 2018-02-08:
Help please, anyone ? 
Please find a link with all of my audio debug info: 

pi@raspberrypi:~ $ cd ~ && wget https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh && bash alsa-info.sh
--2018-02-08 19:56:14--  https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.60.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.60.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27872 (27K) [text/plain]
Saving to: ‘alsa-info.sh’

alsa-info.sh              100%[===================================>]  27.22K  --.-KB/s    in 0.02s

2018-02-08 19:56:14 (1.15 MB/s) - ‘alsa-info.sh’ saved [27872/27872]

ALSA Information Script v 0.4.64
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  lsmod
  aplay
  amixer
  alsactl
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See 'alsa-info.sh --help' for command line options.

cat: '/sys/module/snd_soc_simple_card/parameters/*': No such file or directory
Automatically upload ALSA information to www.alsa-project.org? [y/N] : y
Uploading information to www.alsa-project.org ...  Done!

Your ALSA information is located at http://www.alsa-project.org/db/?f=e674b5a08ee036f10497e03a3eeeff209f35e994

jabulonboaz, 2018-02-08:
This in the URL link bellow please find the detailed debug information, i cant get this to work :(

Your ALSA information is located at http://www.alsa-project.org/db/?f=37022a4253e49bbbc09120eb9411cbe71b51c9f8

pi@raspberrypi:~ $ cd ~ && wget https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh && bash alsa-info.sh
--2018-02-08 20:07:28--  https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.60.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.60.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27872 (27K) [text/plain]
Saving to: ‘alsa-info.sh.1’

alsa-info.sh.1      100%[===================>]  27.22K  --.-KB/s    in 0.03s

2018-02-08 20:07:29 (865 KB/s) - ‘alsa-info.sh.1’ saved [27872/27872]

ALSA Information Script v 0.4.64
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  lsmod
  aplay
  amixer
  alsactl
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See 'alsa-info.sh --help' for command line options.

cat: '/sys/module/snd_soc_simple_card/parameters/*': No such file or directory
Automatically upload ALSA information to www.alsa-project.org? [y/N] : y
Uploading information to www.alsa-project.org ...  Done!

Your ALSA information is located at http://www.alsa-project.org/db/?f=37022a4253e49bbbc09120eb9411cbe71b51c9f8

Pillar1989, 2018-02-09:
@jabulonboaz 
The squeaking wheel gets the oil.
You win.
That's page for you. https://github.com/respeaker/seeed-voicecard/wiki/AlexaPi

jabulonboaz, 2018-02-09:
@Pillar1989 you fixrd it? i will try it immediately!!  
One question it will work on the latest Raspbian jassie lite OS - (no desktop GUI) right?  
Thank you!

jabulonboaz, 2018-02-09:
@Pillar1989 same issue this is not working nothing gets installed this way like i mentioned before :

pi@raspberrypi:~ $ aplay -l
**** List of PLAYBACK Hardware Devices ****
card 0: ALSA [bcm2835 ALSA], device 0: bcm2835 ALSA [bcm2835 ALSA]
  Subdevices: 8/8
  Subdevice #0: subdevice #0
  Subdevice #1: subdevice #1
  Subdevice #2: subdevice #2
  Subdevice #3: subdevice #3
  Subdevice #4: subdevice #4
  Subdevice #5: subdevice #5
  Subdevice #6: subdevice #6
  Subdevice #7: subdevice #7
card 0: ALSA [bcm2835 ALSA], device 1: bcm2835 ALSA [bcm2835 IEC958/HDMI]
  Subdevices: 1/1
  Subdevice #0: subdevice #0

jabulonboaz, 2018-02-10:
Big thank you to @Pilar1989 that help me today fix this. Standing by for the explanation but basically the reapeaker 2mic new drivers would not install due to a parameter that is needed at the repi-confi wrt I2C.   
Stand by for Pillar1989 explanation.

jabulonboaz, 2018-02-10:
Now i am trying to make the respeaker 2mic LEDs to come on when alexa is listening and when she responds.

jabulonboaz, 2018-02-10:
Keeping this open till I can post the respi-config settings that @Pillar1989 changed for me and enabled the new driver to instal

jabulonboaz, 2018-02-10:
Big thanks to @Pillar1989 for finding in the issue and fixing it for me. Apparently the new drivers would not install due to an i2c configuration or enable issue that neded to be performed in raspi-config. Standing by for @Pillar1989 to post here the options that he changed for others that might run into the same issue to beniefit from too.

jabulonboaz, 2018-02-11:
alexa is broken after i tried to activate SPIFeb 11 17:17:38 raspberrypi python[470]: ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Feb 11 17:17:39 raspberrypi python[470]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Feb 11 17:20:06 raspberrypi python[470]: INFO: Checking Internet Connection ...
Feb 11 17:20:07 raspberrypi python[470]: INFO: Connection OK
Feb 11 17:20:07 raspberrypi python[470]: INFO: AVS token: Requesting a new one
Feb 11 17:20:07 raspberrypi python[470]: INFO: AVS token: Obtained successfully
Feb 11 17:20:17 raspberrypi python[470]: INFO: Triggered: pocketsphinx
Feb 11 17:20:21 raspberrypi python[470]: INFO: (process_response Error) Status Code: 403
Feb 11 17:20:23 raspberrypi python[470]: INFO: Triggered: pocketsphinx
Feb 11 17:20:31 raspberrypi python[470]: INFO: (process_response Error) Status Code: 403

DrAlbertWily, 2018-02-15:
@jabulonboaz  If you could post this fix, that would be great. I'm having the same issue. Audio was working fine. I installed Alexa. I now have no audio output. I'm also using the 2 mics hat.

jabulonboaz, 2018-02-15:
@DrAlbertWily https://github.com/respeaker/seeed-voicecard/issues/30

closed 2018-02-15 by jabulonboaz