alexa-pi/AlexaPi #374: AlexaPi service crashing on startup while reading config

craig1971, 2018-02-10:
**Your OS (including version) where you are running AlexaPi:**
```Debian_9.3

**Your hardware platform and model you are running on:**
```OrangePi PC

**Python release (`python2 --version`):**
```2.7.13

**Description of problem:**
```AlexaPi service crashing on startup.  
Appears to be a problem reading the Alexa configuration.
The auth_web.py put the refresh_token in double quotes
I have also tried running without those quotes - the outcome is the same
...
File "/opt/AlexaPi/src/main.py", line 584, in <module>
    token = Token(config['alexa'])
  File "/opt/AlexaPi/src/main.py", line 234, in __init__
    if not self._aconfig.get('refresh_token'):
AttributeError: 'str' object has no attribute 'get'

**Expected:**
``` no crashing

**Problem-relevant `config.yaml` entries:**
...
# Amazon Alexa settings
alexa:
  Client_ID:amzn1.application-oa2-client.84bcaf5xxxxxxxxxxxxxxxxxxxxxxxxxxxxx
  Client_Secret:f4020168d010xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx

  Device_Type_ID:OrangeEcho
  Security_Profile_Description:OrangePiPC
  Security_Profile_ID:amzn1.application.4d55fa2484764cf597033be31a7b4c72
  refresh_token:"Atzr|IwEBIJu1t4rC51Xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

# See https://github.com/alexa-pi/AlexaPi/wiki/Triggers for the description of these options
...

**Steps to reproduce:**

1. run /opt/AlexaPi/src/main.py -d
2. watch it crash
3. cry

**Full Output of Startup in Debug mode**
root@orangepipc:/opt/AlexaPi/src# /opt/AlexaPi/src/main.py -d 
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.H3_Audio_Codec.pcm.front.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM front
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.rear
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.center_lfe
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM cards.pcm.side
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.H3_Audio_Codec.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.H3_Audio_Codec.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround21
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.H3_Audio_Codec.pcm.surround40.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround40
ALSA lib confmisc.c:1281:(snd_func_refer) Unable to find definition 'cards.H3_Audio_Codec.pcm.surround51.0:CARD=0'
ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or directory
ALSA lib conf.c:5007:(snd_config_expand) Evaluate error: No such file or directory
ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM surround41
[... 173 lines cut ...]
INFO: dict.c(320): Allocating 138824 * 20 bytes (2711 KiB) for word entries
INFO: dict.c(333): Reading main dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/cmudict-en-us.dict
INFO: dict.c(213): Dictionary size 134723, allocated 1016 KiB for strings, 1679 KiB for phones
INFO: dict.c(336): 134723 words read
INFO: dict.c(358): Reading filler dictionary: /usr/local/lib/python2.7/dist-packages/pocketsphinx/model/en-us/noisedict
INFO: dict.c(213): Dictionary size 134728, allocated 0 KiB for strings, 0 KiB for phones
INFO: dict.c(361): 5 words read
INFO: dict2pid.c(396): Building PID tables for dictionary
INFO: dict2pid.c(406): Allocating 42^3 * 2 bytes (144 KiB) for word-initial triphones
INFO: dict2pid.c(132): Allocated 21336 bytes (20 KiB) for word-final triphones
INFO: dict2pid.c(196): Allocated 21336 bytes (20 KiB) for single-phone word triphones
INFO: kws_search.c(406): KWS(beam: -1080, plp: -23, default threshold -225, delay 10)
2018-02-10 07:23:46 DEBUG: Setting up playback handler: VlcHandler
2018-02-10 07:23:46 DEBUG: setup
2018-02-10 07:23:46 INFO: Checking Internet Connection ...
2018-02-10 07:23:46 DEBUG: Starting new HTTPS connection (1): api.amazon.com
2018-02-10 07:23:46 DEBUG: https://api.amazon.com:443 "GET /auth/o2/token HTTP/1.1" 404 29
2018-02-10 07:23:46 INFO: Connection OK
Traceback (most recent call last):
  File "/opt/AlexaPi/src/main.py", line 584, in <module>
    token = Token(config['alexa'])
  File "/opt/AlexaPi/src/main.py", line 234, in __init__
    if not self._aconfig.get('refresh_token'):
AttributeError: 'str' object has no attribute 'get'
root@orangepipc:

gunnicus, 2018-02-27:
I had this issue, after debugging, I figured out in the config.yaml file, you need to add a white space in front of the value, like below:

[image]

EmerickH, 2018-07-19:
Due to a large number of issues, this one is going to be closed because we think your problem is solved.
If you still have any problem, please feel free to re-open it!

closed 2018-07-19 by EmerickH