alexa-pi/AlexaPi #375: How to trigger the top 3 rob leds of respeaker 2mic hat on a RaspberryPi zero w with an AlexaPi build when Alexa listens and responds?

jabulonboaz, 2018-02-10:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Raspbian Jessie lite
```

**Your hardware platform and model you are running on:**
Pi zero w and pi3 vb

**Python release (`python2 --version`):**
```
Do not know
```

**Description of problem:**
How to trigger the top 3 rob leds of respeaker 2mic hat on a RaspberryPi zero w with an AlexaPi build when Alexa listens and responds?

**Expected:**
Control the respeaker 2 mic hat hat 3 rgb leds when Alexa listens and talks 
**Problem-relevant `config.yaml` entries:**
GPIO config on rspberry pi

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
Does anyone know how the top 3 leds of respeaker 2mic hat be triggered when Alexa listens and talks on a RaspberryPi zero w with an AlexaPi build please?

renekliment, 2018-02-10:
If the LEDs are connected directly to RPi's GPIO pins, you can use the existing `raspberrypi` _device platform_ and set that up in the configuration file, it's pretty straightforward.

If you need something more complex, feel free to write your own _[device platform file](https://github.com/alexa-pi/AlexaPi/tree/master/src/alexapi/device_platforms)_.

jabulonboaz, 2018-02-11:
@renekliment they are not unfortunately, this is from their product description ; I dot try to change the GPIO settings in the AlexaPi relevant raspberry pi LED section and I did successfully enabled the Respeaker button on GPIO 17 for Alexa triggering by pressing it in addition to the hotword activation but when I change the LED settings by inserting GPIO 10 that is the respeaker pin for its rgb leds nothing happens. 
‘RGB LED: 3 APA102 RGB LEDs, connected to SPI interface’
I am just an armature with an engineering background trying to learn by googling and reading the great work that yourself and others are putting together I am not a coder or a developer I am afraid .
Any pointers, any URLs or any tips you can recommend I am happy to follow. 
Another thing I like to do is to be able to control with buttoyyns connected to GPIO pins the volume of Alsa mixer again any tips and online material you can point too I would be very grateful.

jabulonboaz, 2018-02-11:
after playing with the SPI Alexa stooped working Feb 11 17:17:38 raspberrypi python[470]: ALSA lib pcm_dmix.c:990:(snd_pcm_dmix_open) The dmix plugin supports only playback stream
Feb 11 17:17:39 raspberrypi python[470]: ALSA lib pcm_dsnoop.c:556:(snd_pcm_dsnoop_open) The dsnoop plugin supports only capture stream
Feb 11 17:20:06 raspberrypi python[470]: INFO: Checking Internet Connection ...
Feb 11 17:20:07 raspberrypi python[470]: INFO: Connection OK
Feb 11 17:20:07 raspberrypi python[470]: INFO: AVS token: Requesting a new one
Feb 11 17:20:07 raspberrypi python[470]: INFO: AVS token: Obtained successfully
Feb 11 17:20:17 raspberrypi python[470]: INFO: Triggered: pocketsphinx
Feb 11 17:20:21 raspberrypi python[470]: INFO: (process_response Error) Status Code: 403
Feb 11 17:20:23 raspberrypi python[470]: INFO: Triggered: pocketsphinx
Feb 11 17:20:31 raspberrypi python[470]: INFO: (process_response Error) Status Code: 403

any ideas ?

EmerickH, 2018-07-19:
You can use `event_commands` in the config to launch your own commands:
https://github.com/alexa-pi/AlexaPi/blob/master/src/config.template.yaml#L80