alexa-pi/AlexaPi #376: Alexa does not answer questions just responds with a "Yes" and then science

jabulonboaz, 2018-02-11:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Raspbian Jessie lite

**Your hardware platform and model you are running on:**
Raspberry Pi zero W + Respeaker 2mic hat 

**Python release (`python2 --version`):**
Python 2.7.13`

**Description of problem:**
Alexa does not answer questions just responds with a "Yes" and then there science.
I can trigger Alexa by my voice and the button but just says yes.
It was all working this morning then this started, I rebuild-ed the entire thing from scratch with a fresh copy of Raspbian just in case and everything in the  AlexaPi installation process concluded fine. 
I can also notice that there is delay between the time you can trigger Alexa after the last trigger it takes like 30 seconds or so for her to be able to respond with just a "Yes" again.

**Expected:**
Alexa to answer to questions etc after triggered e.g.  Q: "Alexa?" A: "Yes" Q:"Whats the time?" A:" the time is ...."

**Problem-relevant `config.yaml` entries:**
Don't know

**Steps to reproduce:**
1.  Install Alexa Pi and followed the process as described in this git  

**Traceback (if applicable):**
I can provide my history command output 

**Additional info:**
these are my logs:
I rebuild-ed everything from scratch over and over, the sound card installed and works fine, Alexa listens and response with yes but never continues to replay to the question i asked. 
She just says yes and then silence, here are my logs, any idea anyone please?

Feb 11 19:30:17 raspberrypi python[474]: INFO: Checking Internet Connection ...
Feb 11 19:30:18 raspberrypi python[474]: INFO: Connection OK
Feb 11 19:30:18 raspberrypi python[474]: INFO: AVS token: Requesting a new one
Feb 11 19:30:18 raspberrypi python[474]: INFO: AVS token: Obtained successfully
Feb 11 19:30:30 raspberrypi python[474]: INFO: Triggered: pocketsphinx
Feb 11 19:30:33 raspberrypi python[474]: INFO: (process_response Error) Status Code: 403
Feb 11 19:30:42 raspberrypi python[474]: INFO: Triggered: pocketsphinx
Feb 11 19:30:46 raspberrypi python[474]: INFO: (process_response Error) Status Code: 403
Feb 11 19:32:19 raspberrypi python[474]: INFO: Triggered: pocketsphinx
Feb 11 19:32:23 raspberrypi python[474]: INFO: (process_response Error) Status Code: 403

pi@raspberrypi:~ $ cd ~ && wget https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh && bash alsa-info.sh
--2018-02-11 19:51:24--  https://raw.githubusercontent.com/alexa-pi/AlexaPi-misc/master/alsa-info.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.60.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.60.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 27872 (27K) [text/plain]
Saving to: ‘alsa-info.sh’

alsa-info.sh                100%[==========================================>]  27.22K  --.-KB/s    in 0.01s

2018-02-11 19:51:25 (2.16 MB/s) - ‘alsa-info.sh’ saved [27872/27872]

ALSA Information Script v 0.4.64
--------------------------------

This script visits the following commands/files to collect diagnostic
information about your ALSA installation and sound related hardware.

  dmesg
  lspci
  lsmod
  aplay
  amixer
  alsactl
  /proc/asound/
  /sys/class/sound/
  ~/.asoundrc (etc.)

See 'alsa-info.sh --help' for command line options.

cat: '/sys/module/snd_soc_simple_card/parameters/*': No such file or directory
Automatically upload ALSA information to www.alsa-project.org? [y/N] : y
Uploading information to www.alsa-project.org ...  Done!

Your ALSA information is located at http://www.alsa-project.org/db/?f=1bfc46ac5afe839401bc592ecf43f70c3b25e93c

Please inform the person helping you.

McrESP, 2018-02-12:
Same problem. Alexa responds with "yes" and then nothing. I'm getting Status Code 403 as well.

futurepr0n, 2018-02-12:
SAme problem recently too! anyone know how to fix - please report here:)

renekliment, 2018-02-12:
I can confirm this.

I couldn't find an official statement from AVS ... maybe they indeed shut down the old API or just changed the auth mechanism a little - how knows.
It throws

```
{
    "error:{
        "code": "com.amazon.alexahttpproxy.exceptions.InvalidAccessTokenException",
        "message": "AVS was not able to validate the provided access token."
    }
}
```

Sadly, I don't have the time to investigate this further. If there are any good PRs fixing this, I'll be happy to merge them.

kodxana, 2018-02-12:
So I see Im not alone with this same problem.

SpectrumTech, 2018-02-12:
I noticed that the origin URL’s info tag says that it should be https and for some weird reason wasn’t wanting to take the http://192.168.x.x:5050 url at all at first. I thought maybe it had to be https, but finally I was able to get it to take. I don’t know enough about web programming to understand what that part is actually doing, I assume letting amazon know where to expect requests to come from and what port? But scanning through the source code I notice everything is https protocol and never saw where http://your.pi.ip.address:5050 is coded into try changing (breaking) it.

jabulonboaz, 2018-02-12:
@SpectrumTech did you try to build AlexaPi today or yesterday with https instead of http and it all worked? Alexa responds with more than just a "yes"?

samfreudiger, 2018-02-12:
having the exact same issue after following this guide: https://lifehacker.com/the-simplest-way-to-build-a-raspberry-pi-powered-amazon-1794218212

SpectrumTech, 2018-02-12:
@jabulonboaz I didn’t try building it with https at all. And no, all it says is “yes” but nothing more. 403 error.

When I went to add the origin URL in the Amazon dev console it would not let me click “add” when I put in the http version per the AlexaPi installation instructions during my first attempt - so I clicked the (?) help button and it said it should be an https address. So I tried that, but it wouldn’t work when I’d finally get to the final validation step. It would give the “we’re sorry” page - I assume because the AlexaPi code is sending an http request and the AVS expected https? I fiddled around a bit and was finally able to get the dev console to let me add an http address, which would allow validation to complete and everything seemed ok, except for the 403 errors.

I did a quick look through the source code for the project, just to see where it is using http://192.168.x.x:5050 and if it could be changed to https, but only saw where it gets printed to the display during setup. I don’t really know that it is even a problem and what would happen if that part was changed. This was my first attempt at using AlexaPi or using the AVS. I only know just enough programming to be dangerous! Someone smarter than me will have to fix this 😊

renekliment has what seems to be the best raspberry pi / Alexa project around, but no time to continue it. I do want to say thanks for all their work on it and best wishes on whatever comes next.  Hopefully someone will pick this beautiful project up and continue it.

vsinitsa, 2018-02-13:
I had the same issue and when I went to my Developer Console and in the security profile I added both http and https links it started responding with answers to my questions

jabulonboaz, 2018-02-13:
@vsinitsa thank you for this i will give it a go.

ycreng, 2018-02-13:
Yes, using both http and https links fixed it.  Thanks!

ycreng, 2018-02-13:
In fact, you only need to have https links.

vsinitsa, 2018-02-13:
@ycreng thats probably true but I had to add the http links to my security profile for the authorization page to not error out one me. So after authorizing you can probably just remove the http links

ycreng, 2018-02-13:
Ah yes, I didn't think about authorization.  Thanks for the correction.

pfiol, 2018-02-13:
@ycreng thanks for the tip! I will try it when I get home. 

So, If I understood correctly, to make it work we just have to:
1 - Go to Security Profile tab and under "Allowed origins" add:
https://localhost:5050/
2 - Under "Allowed return URLs" add:
https://localhost:5050/code

Is that correct? I guess afterwards we have to restart AlexaPi service as well.

samfreudiger, 2018-02-13:
yeah!
for me it started working as soon as i removed all http:// URLs

jabulonboaz, 2018-02-13:
Yes i can confirm is working no need to reprovision from scratch just go to the developer portal and add the same options with https too. However Alexa is not listening with the same sensitivity as before and i check alsamixer levels they have not changed. In many occasions she won't listen at all so no hello. The button works fine though. 
Maybe is my sound card and mics but it was all working flawlessly before the https issue.

ycreng, 2018-02-13:
@pfiol As vsinitsa (who found the fix) points out, you will need to have http links in order to get through the initial authorization step.  After you have been authorized, you can use just https links.  Your understanding is correct with respect to the https links for "Allowed origins" and "Allowed return URLs".  I have both http and https links, and everything works fine.

pfiol, 2018-02-14:
@ycreng Absolutely correct. Tested and working.

The wiki should be updated for new users to take this change into account.

jabulonboaz, 2018-02-15:
i do not know where i can ask this question so i place it here and please feel free to point me to the right direction.

How can i point Alexa to run a script i have i have  please?
E.g "hey Alexa trigger wifi on" and then a wifi_on.sh script is executed.

closed 2018-02-15 by jabulonboaz