alexa-pi/AlexaPi #377: Permission Denied to Config, along with response error "Yes" and then flatlining.

aaronvejo, 2018-02-15:
**Your OS (including version) where you are running AlexaPi:**
```NOOVA 2.4.5, Rasbian
```

**Your hardware platform and model you are running on:**
```Raspberry Pi 3 

```

**Python release (`python2 --version`):**
```Python 2.7

```

**Description of problem:**
``'Only prompted response during debugging and not while running when status is active. More often than not the only response is a "Yes" and then nothing in the debugging more. While trying to figure out the issue I can not access the configurations page it just prompts "permission denied". Any advice?

```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```config entries restricted

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```
Here is the status update when running considering I can't access the configs

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-02-15 18:48:40 UTC; 40s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 2029 (python)
   CGroup: /system.slice/AlexaPi.service
           ├─2029 /usr/bin/python /opt/AlexaPi/src/main.py --daemon
           └─2041 /usr/bin/pulseaudio --start --log-target=syslog

Feb 15 18:48:44 raspberrypi python[2029]: ALSA lib conf.c:4528:(_snd_config_evaluate) function snd_func_refer returned error: No such file or d
Feb 15 18:48:44 raspberrypi python[2029]: ALSA lib conf.c:4996:(snd_config_expand) Args evaluate error: No such file or directory
Feb 15 18:48:44 raspberrypi python[2029]: ALSA lib pcm.c:2495:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Feb 15 18:48:44 raspberrypi python[2029]: ALSA lib pcm_dmix.c:1052:(snd_pcm_dmix_open) unable to open slave
Feb 15 18:48:44 raspberrypi python[2029]: connect(2) call to /tmp/jack-999/default/jack_0 failed (err=No such file or directory)
Feb 15 18:48:44 raspberrypi python[2029]: attempt to connect to server failed
Feb 15 18:48:45 raspberrypi python[2029]: INFO: Checking Internet Connection ...
Feb 15 18:48:45 raspberrypi python[2029]: INFO: Connection OK
Feb 15 18:48:45 raspberrypi python[2029]: INFO: AVS token: Requesting a new one
Feb 15 18:48:46 raspberrypi python[2029]: INFO: AVS token: Obtained successfully

```

vsinitsa, 2018-02-15:
In your security profile on the amazon developer page add both http and https links into your allowed origins and allowed return urls. Http links are needed for authentication while https links are needed for alexa to not just ignore your commands

aaronvejo, 2018-02-15:
Thanks, that solved one of the problems. I'm no longer getting the errors in my status but she still won't respond when actively and I can still not access my configuration file. Any suggestions?

EmerickH, 2018-07-19:
You have to setup your audio PCMs:
List them with `aplay -L` and change it in the config:
```
output: "alsa"
output_device: "plughw:YOURDEVICE"
```
On Rapsberry Pi, you may have to follow [this](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#running-pa-in-system-wide-mode-recommended) (with SoX, not VLC at the end)

EmerickH, 2018-07-19:
Due to a large number of issues, this one is going to be closed because we think your problem is solved.
If you still have any problem, please feel free to re-open it!

closed 2018-07-19 by EmerickH