alexa-pi/AlexaPi #378: How can i make Alexa to trogger/execute a script .sh file i have in my Pi with a custom trigger word please?

jabulonboaz, 2018-02-15:
How can i make Alexa to trogger/execute a script .sh file i have in my Pi with a custom trigger word please?
E.g "Alexa trigger wifi on" and then a wifi_on.sh script is executed from within my Pi.

jabulonboaz, 2018-02-17:
Anyone pls?

renekliment, 2018-02-17:
Since this is pretty frequently asked, it's now in the F.A.Q.: https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#how-do-i-run-a-local-command--script-when-i-say-x

closed 2018-02-17 by renekliment

jabulonboaz, 2018-02-17:
@renekliment excuse my ignorance but that url you just posted points to an empty uncreated FAQ with no question or answer. Did i do something wrong?
Have you posted an answer somewhere?
Regards

renekliment, 2018-02-17:
@jabulonboaz The link works. Make sure you click it so you don't miss something in the URL. You can also get there from local wiki. Post a screenshot if your issue persists.

jabulonboaz, 2018-02-17:
@renekliment thank u i found it now but the answer is very disappointing as i lack the skills to do number 2 and even though i know how to write my own skill and have done so in the past to do option 1 that won't work for my original ask which is local integration as in everything is triggered from the Pi HW behind the same LAN.