alexa-pi/AlexaPi #384: Failing on Install

gschwiter, 2018-03-09:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
Linux Version 4.4.44-1-ARCH
HamVoip 1.5rc4-15 AllStar
Asterisk 1.4.23

**Your hardware platform and model you are running on:**
PI3

**Python release (`python2 --version`):**
I think the problem is here
python2-setuptools: /usr/bin/easy_install-2.7 exists in filesystem

**Description of problem:**
I am having an issue with the initial install, I believe its failing on the following.
error: failed to commit transaction (conflicting files)
python2-setuptools: /usr/bin/easy_install-2.7 exists in filesystem
Errors occurred, no packages were upgraded.
[root@node1 opt]#

**Additional info:**

Failing on Install
I am having an issue with the initial install, I believe its failing on the following.
error: failed to commit transaction (conflicting files)
python2-setuptools: /usr/bin/easy_install-2.7 exists in filesystem
Errors occurred, no packages were upgraded.
[root@node1 opt]#

I’m trying to run this on an existing build to integrate it into another project. Listed is the version of linux and other packages running on the PI3.
Linux Version 4.4.44-1-ARCH
HamVoip 1.5rc4-15 AllStar
Asterisk 1.4.23
I have listed the full setup errors. Any ides or direction would be great.
 root@node1 opt]# sudo pacman -Sy git
:: Synchronizing package databases...
 core is up to date
 extra is up to date
 community is up to date
 alarm is up to date
 aur is up to date
 hamvoip is up to date
warning: git-2.7.4-1 is up to date -- reinstalling
resolving dependencies...
looking for conflicting packages...

Packages (1) git-2.7.4-1

Total Download Size:    3.46 MiB
Total Installed Size:  23.56 MiB
Net Upgrade Size:       0.00 MiB

:: Proceed with installation? [Y/n] y
:: Retrieving packages...
 git-2.7.4-1-armv7h                    3.5 MiB  2.23M/s 00:02 [##################################] 100%
(1/1) checking keys in keyring                                [##################################] 100%
(1/1) checking package integrity                              [##################################] 100%
(1/1) loading package files                                   [##################################] 100%
(1/1) checking for file conflicts                             [##################################] 100%
(1/1) checking available disk space                           [##################################] 100%
:: Processing package changes...
[... 153 lines cut ...]
(123/123) checking for file conflicts                         [##################################] 100%
error: failed to commit transaction (conflicting files)
python2-setuptools: /usr/bin/easy_install-2.7 exists in filesystem
Errors occurred, no packages were upgraded.
[root@node1 opt]# ^C
[root@node1 opt]#

[root@node1 opt]# sudo systemctl status AlexaPi.service
* AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: disabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           `-usergroup-root.conf
   Active: failed (Result: exit-code) since Thu 2018-03-08 20:44:56 HST; 27s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 1825 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAIL
 Main PID: 1825 (code=exited, status=1/FAILURE)

Mar 08 20:44:55 node1 systemd[1]: Started Alexa client for all your devices.
Mar 08 20:44:56 node1 python[1825]: Traceback (most recent call last):
Mar 08 20:44:56 node1 python[1825]:   File "/opt/AlexaPi/src/main.py", line 17, in <module>
Mar 08 20:44:56 node1 python[1825]:     from future.builtins import bytes
Mar 08 20:44:56 node1 python[1825]: ImportError: No module named 'future'
Mar 08 20:44:56 node1 systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 20:44:56 node1 systemd[1]: AlexaPi.service: Unit entered failed state.
Mar 08 20:44:56 node1 systemd[1]: AlexaPi.service: Failed with result 'exit-code'.

maso27, 2018-03-09:
You could try running
`pacman -S --force python2 python2-pip`
before the install.

Once they're installed, it shouldn't try to install them again in the script.

I'm basing that on the suggestion [here](https://stackoverflow.com/questions/38286602/archlinux-pip-and-setuptools-not-upgraded), so proceed at your own risk.

gschwiter, 2018-03-10:
[root@node1 opt]# sudo systemctl status AlexaPi.service
* AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: disabled)
  Drop-In: /etc/systemd/system/AlexaPi.service.d
           `-force-python2.conf, usergroup-root.conf
   Active: active (running) since Fri 2018-03-09 21:35:32 HST; 2h 19min ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
 Main PID: 369 (python2)
    Tasks: 3 (limit: 512)
   CGroup: /system.slice/AlexaPi.service
           `-369 /usr/bin/python2 /opt/AlexaPi/src/main.py --daemon

Mar 09 21:35:42 node1 python2[369]:   File "/usr/lib/python2.7/site-packages/pyaudio.py", line 441, in __init__
Mar 09 21:35:42 node1 python2[369]:     self._stream = pa.open(**arguments)
Mar 09 21:35:42 node1 python2[369]: IOError: [Errno -9996] Invalid input device (no default output device)
Mar 09 21:35:42 node1 python2[369]: ALSA lib pcm_dmix.c:1029:(snd_pcm_dmix_open) unable to open slave
Mar 09 21:35:42 node1 python2[369]: [73d00f00] alsa audio output error: cannot open ALSA device "default": Device or resource busy
Mar 09 21:35:42 node1 python2[369]: [73d00f00] core audio output error: Audio output failed
Mar 09 21:35:42 node1 python2[369]: [73d00f00] core audio output error: The audio device "default" could not be used:
Mar 09 21:35:42 node1 python2[369]: Device or resource busy.
Mar 09 21:35:42 node1 python2[369]: [73d00f00] core audio output error: module not functional
Mar 09 21:35:42 node1 python2[369]: [73924ab0] core decoder error: failed to create audio output

EmerickH, 2018-07-19:
Hello,
you must setup your audio
List output devices with `arecord -L` and change it in the config with something like:
```
input_device = plughw:CARD=YOURCARD,DEV=0
```
(you may also have to do it with `output_device ` and `aplay -L`)

EmerickH, 2018-07-19:
Due to a large number of issues, this one is going to be closed because we think your problem is solved.
If you still have any problem, please feel free to re-open it!

closed 2018-07-19 by EmerickH