alexa-pi/AlexaPi #387: vlc-nox

vc551, 2018-03-13:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**
```
ubuntu 17.10
```

**Your hardware platform and model you are running on:**
```
desktop-laptop toshiba portege m800
```

**Python release (`python2 --version`):**
```

```

**Description of problem:**
```
When I start setup.sh, it downloads the prerequisites after asking whether airplay support should be added or not. It then terminates giving the error- unable to locate package vlc-nox
```

**Expected:**
```

```

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**
```

```

iLuke2210, 2018-03-31:
Same issue here

EmerickH, 2018-07-19:
`vlc-nox` doesn't exists in Ubuntu 17.04, see:
https://askubuntu.com/questions/1028728/unable-to-install-vlc-nox

Not an AlexaPi bug

closed 2018-07-19 by EmerickH

jolimk, 2019-07-08:
Hey, im having the same issue. How did you solve it?

Kleinstein-github, 2019-07-28:
Here is the FIX:

https://richardtech.net/2019/03/alexa-on-raspberry-pi/