alexa-pi/AlexaPi #389: Amazon Trust Services support?

aarmea, 2018-03-27:
This is more of a question than an actual issue, but I recently received this email from Amazon claiming that my AVS client needs to support certificates from ATS before June 15:

[image]

Is this something I need to worry about for my AlexaPi device? If ATS support is already implemented, please close this issue.

Thanks,
Albert

futurepr0n, 2018-04-11:
I have the same question. I noticed lately I can't get Alexa to turn on when my Pi has rebooted... The service, when started manually doesnt seem to actually turn on either. Wondering if this is related to the issue?

EmerickH, 2018-07-19:
I don't think ATS is implemented in AlexaPi...
No active development is made on it so consider switching to [a project like this one](https://github.com/alexa/alexa-avs-sample-app) for example (actually in maintenance but you can find another)