alexa-pi/AlexaPi #391: What is possible with AlexaPi ?

howi42, 2018-04-01:
I have AlexaPi up and running, asking questions and getting a result works fine.
But what now ?
Is there anything I can connect to AlexaPi ?
Example : 
- a calendar to get better answers about specific dates ?
- voice control commands to control Alexa enabled devices ?

smowk, 2018-04-03:
you can add skills to your Alexa and use it as you want.

Let's say you have a Philips HUE; you can add the PhilipsHue skill to Alexa and when you say Alexa/ Turn lights off it will turn off the lights for you

maso27, 2018-04-03:
@smowk, have you had this work for you, using AlexaPi?

To my knowledge, the smart home connection stuff is implemented in the various Alexa devices' internal software, not at the server.
AlexaPi is merely a window to the server--sending and receiving audio files.

AlexaPi works great with any skills that are strictly audio-based but we just don't have access to or support for the client-side smart devices controls.

I feel like I heard reports of folks who also had echo devices connected being able to control their smart devices with AlexaPi but I'm not certain of it.

EmerickH, 2018-07-19:
@fourleggedostrich no

AlexaPi is using Alexa API v1 that is very limited so a lot of skills can never work on it.
I leave this issue openned to list skills that are working or not