alexa-pi/AlexaPi #395: 500 internal server Error

jbeechey, 2018-04-15:
**Your hardware platform and model you are running on:**
Raspberry Pi 3b+ 

**Description of problem:**
I am attempting to run AlexaPi with MagicMirror. I have MM always installed and running. 
Followed this guide for AlexaPi
https://pchelp.ricmedia.com/build-alexapi-with-amazon-alexa-raspberry-pi/3/
I was able to log in to the amazon page, but then
I received an 500 internal error. I doubled checked the config.yml and the auth file and everything seems to match up. I attempted to uninstall and reinstall with no luck. Any help appreciated. I am a NOOB, so please guide me through this.

6bowens, 2018-04-18:
I'm having the same issue on RPi 2b+

Installation script ran well until auth_web.py

Resulted in 500 Internal Server error

dvbit, 2018-04-24:
Check very well how you copy ids . For me it was a matter of cut and paste

EmerickH, 2018-07-19:
Logging error are 99% of the time caused by wrong API keys, device IDs, etc..
Please check you have well entered them.
([tutorial](https://github.com/alexa-pi/AlexaPi/wiki/Installation#1-register-at-amazon))
I close this issue, please feel free to re-open it if you still have a problem.

closed 2018-07-19 by EmerickH