alexa-pi/AlexaPi #396: Authorisation Error

ewimp1, 2018-04-23:
Hello, I am following the steps in

https://lifehacker.com/the-simplest-way-to-build-a-raspberry-pi-powered-amazon-1794218212

to use my Raspberry Pi as an Echo dot.

Currently I am unable to authorise my devise: when I go to http://192.168.1.105:5050 I get the error:

The redirect URI you provided has not been whitelisted for your application. Please add your redirect URI in the 'Allowed Return URLs' section under 'Web Settings' for your Security Profile on Amazon Developer Portal.

In my security profile on my Amazon developers account I have:

Allowed Origins | http://localhost:5050 | https://localhost:5050 | https://192.168.1.105:5050
Allowed Return URLs https://localhost:5050/code | http://localhost:5050/code | https://192.168.1.105:5050/code | http://localhost:5050/authresponse | https://localhost:5050/authresponse | https://192.168.1.105:5050/authresponse | https://192.168.1.105/code | https://192.168.1.105/authresponse

I can't add the Raspberry Pi IP address without https. The port 5050 is the one I have been asked to go to when running sudo ./AlexaPi/src/scripts/setup.sh.

Also just to mention, I have tried to go to http://192.168.1.105:5050 in my laptop's browser. I am running the installation to the Pi through my laptop by SSH.

If anyone can help that would be great!

PS I am a complete newbie so assume I know nothing and simple answers would be much appreciated.

ewimp1, 2018-04-23:
I asked the same question here

https://github.com/alexa/alexa-avs-sample-app/issues/1173

 but now I am thinking I would also like a response directly related to  AlexaPi.

Thanks!

maso27, 2018-04-23:
I haven't run an AlexaPi in a while, but I'm pretty sure this have changed on the server side.
Try going to https instead of http and see what happens.

susatenk, 2018-05-01:
Indeed, now http allowed only for localhost address.  In order to run HTTPS server you need to: 
1) Generate SSL certificate.
#  openssl genrsa -out privkey.pem 2048
#  openssl req -new -x509 -days 7300 -key privkey.pem -out cert.pem

2) Modify script /opt/AlexaPi/src/auth_web.py to use it.
================
cherrypy.server.ssl_module = 'builtin'
cherrypy.server.ssl_certificate = "/opt/AlexaPi/ssl/cert.pem"
cherrypy.server.ssl_private_key = "/opt/AlexaPi/ssl/privkey.pem"
================
Put these three lines before "cherrypy.quickstart(Start())" and don't forget to fix paths to your actual certificate and key.

EmerickH, 2018-07-19:
Solved in:
https://github.com/alexa/alexa-avs-sample-app/issues/1173#issuecomment-384094255

closed 2018-07-19 by EmerickH