alexa-pi/AlexaPi #397: Issue with Token (code=exited, status=1/FAILURE)**Expected

mikeb8319, 2018-04-26:
Thank you for reporting an issue with **AlexaPi**.

Make sure you are running the latest version of **AlexaPi**, and look through [issues](https://github.com/alexa-pi/AlexaPi/issues?utf8=%E2%9C%93&q=is%3Aissue) before submitting.

Check out our [Wiki](https://github.com/alexa-pi/AlexaPi/wiki/), and make sure to look at [Debugging](https://github.com/alexa-pi/AlexaPi/wiki/Debugging/), and [Audio debugging](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging) in particular before filing an issue.

In order to help troubleshooting, be sure to include the following information:

**Output of the [audio debugging script](https://github.com/alexa-pi/AlexaPi/wiki/Audio-setup-&-debugging#audio-debugging-script).**

**Your OS (including version) where you are running AlexaPi:**

PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"

**Your hardware platform and model you are running on:**
Raspberry pi 3 

**Python release (`python2 --version`):**
Python 2.7.13

**Description of problem:**
 
My Alexa Pi is not starting, and I believe its due to not being able to get the token at the end of the install. I'm not sure what "(code=exited, status=1/FAILURE)**Expected:**" and tried looking it up on the forums with no luck. Does anyone know how to fix this issue?

**Problem-relevant `config.yaml` entries:**
```

```

**Steps to reproduce:**

1. 
2. 
3. 

**Traceback (if applicable):**
```

```

**Additional info:**

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-04-26 10:03:43 EDT; 1min 6s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 5165 ExecStart=/usr/bin/python /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 5165 (code=exited, status=1/FAILURE)

Apr 26 10:03:43 raspberrypi python[5165]:   File "/opt/AlexaPi/src/alexapi/triggers/__init__.py", line 10, in init
Apr 26 10:03:43 raspberrypi python[5165]:     im = importlib.import_module('alexapi.triggers.' + name + 'trigger', 
Apr 26 10:03:43 raspberrypi python[5165]:   File "/usr/lib/python2.7/importlib/__init__.py", line 37, in import_mod
Apr 26 10:03:43 raspberrypi python[5165]:     __import__(name)
Apr 26 10:03:43 raspberrypi python[5165]:   File "/opt/AlexaPi/src/alexapi/triggers/pocketsphinxtrigger.py", line 6
Apr 26 10:03:43 raspberrypi python[5165]:     from pocketsphinx import get_model_path
Apr 26 10:03:43 raspberrypi python[5165]: ImportError: cannot import name get_model_path
Apr 26 10:03:43 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Apr 26 10:03:43 raspberrypi systemd[1]: AlexaPi.service: Unit entered failed state.
Apr 26 10:03:43 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.

BullionMan, 2018-05-31:
I'm seeing the exact same output here. Any news on this out there?

EmerickH, 2018-07-19:
There is 2 diferents error and the second one (that we can see in the log) seems to be cause by a faulty install of PocketSphinx (like #165).
Try to reinstall it:
```
pip install --upgrade --force-reinstall pocketsphinx
```
For the first bug, without much precisions I can't help you...

EmerickH, 2019-01-07:
@BullionMan and @mikeb8319 You don't replied so I suppose that your problem is fixed.
If you still have the issue, please re-open this issue and provide more information please.

Emerick

closed 2019-01-07 by EmerickH

happy2behere, 2020-04-16:
hi and thank you for everyone help i do appreciate it if no one else does.
ive tryed  pip install --upgrade --force-reinstall pocketspinx but this did not work
so after running this 

sudo systemctl status AlexaPi.service 

i get this 

● AlexaPi.service - Alexa client for all your devices
   Loaded: loaded (/usr/lib/systemd/system/AlexaPi.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2020-04-16 17:31:26 BST; 1s ago
     Docs: https://github.com/alexa-pi/AlexaPi/wiki
  Process: 2276 ExecStart=/usr/bin/python3 /opt/AlexaPi/src/main.py --daemon (code=exited, status=1/FAILURE)
 Main PID: 2276 (code=exited, status=1/FAILURE)

Apr 16 17:31:22 raspberrypi python3[2276]: ALSA lib conf.c:5036:(snd_config_expand) Args evaluate error: No such file or directory
Apr 16 17:31:22 raspberrypi python3[2276]: ALSA lib pcm.c:2565:(snd_pcm_open_noupdate) Unknown PCM bluealsa
Apr 16 17:31:22 raspberrypi python3[2276]: Expression 'alsa_snd_pcm_hw_params_set_period_size_near( pcm, hwParams, &alsaPeriodFrames, &dir )' failed in 'src/hostapi/alsa/pa_linux_alsa.c', line: 924
Apr 16 17:31:24 raspberrypi python3[2276]: /opt/AlexaPi/src/main.py:71: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full d
Apr 16 17:31:24 raspberrypi python3[2276]:   config = yaml.load(stream)
Apr 16 17:31:24 raspberrypi python3[2276]: INFO: Checking Internet Connection ...
Apr 16 17:31:25 raspberrypi python3[2276]: INFO: Connection OK
Apr 16 17:31:25 raspberrypi python3[2276]: CRITICAL: AVS refresh_token not found in the configuration file. Run the setup again to fix your installation (see project wiki for installation instructions).
Apr 16 17:31:26 raspberrypi systemd[1]: AlexaPi.service: Main process exited, code=exited, status=1/FAILURE
Apr 16 17:31:26 raspberrypi systemd[1]: AlexaPi.service: Failed with result 'exit-code'.

any help will be much appreciated and thank you in advance