alexa-pi/AlexaPi #403: How do i change the language to en_gb language?

malikasinger1, 2018-06-03:
I just need to figure out how to get AlexaPi to work with the EN_gb language.  I tried changing my location but that didn't work

It is all about accent, canni even change accent of the AlexaPi

Looking forward for your help
Thanks in advance

Inzamam Malik
Chatbot developer

cward1299, 2018-06-27:
Hi Inzamam,

I am after the exact same as you but cant find this information anywhere. Dont suppose you figured it out?

malikasinger1, 2018-06-28:
Not yet

EmerickH, 2018-07-19:
Hello,
Like renekliment said in #328, changing language is not supported at this time because AlexaPi is using AVS API v1 (first version made for USA only).
AlexaPi has no active developpement, maybe you can find another project that is working with new API.

Emerick

EmerickH, 2018-07-30:
Sorry, it is now suported ([Q&A](https://github.com/alexa-pi/AlexaPi/wiki/Q&A-(FAQ)#can-i-switch-the-language-to-german-etc))!
You just have to put your Amazon Developper account in your language.

closed 2018-07-30 by EmerickH